aredn/aredn_ar71xx #3: Code is diverging from official AREDN code

cmlara, 2018-04-16:
The code in this repo is diverging from the official AREDN(TM) repository available via git://git.aredn.org and code review server (sorry I can't give the URL for code review, I've requested for years to make code review public to all members of the community but every time I ask I've been turned down by the group that calls its self the core team.)

This appears to be an unofficial repo not subject to AREDN(TM) code review standards.

Please remove the affiliation with the AREDN(TM) project as required by the  license file and make it clear that this is not an AREDN project.

wu2s, 2018-04-16:
The AREDN(TM) trademark and AREDN(TM) logo are property of Randolph Smith. As the trademark holder, I am authorizing this official test repository for AREDN(TM). Where questions arise as to the licensed use or the validity of an AREDN(TM) software release, the decision of Randolph Smith shall be binding.

closed 2018-04-16 by wu2s

cmlara, 2018-04-16:
The AREDN(TM) Character Mark and Amateur Radio Emergency Data Network name however are property of Conrad Marcus Lara KG6JEI (myself) so there is a concern there that he has some say in the matter of the use and you only control the graphical image which is heavily based on the base intellectual property it was modeled after.

Since you control the Trademark Logo graphic (a really small part of the project) are you now officially stating that github is now the single official source of code that can use your image or are two competing repositories with diverged code paths?

Please advise so I can ensure the repositories I am responsible for are kept up to date with your current contract wishes to your logo, if you have moved to github only for your use of your image I will ensure we remove your logo file from the git.aredn.org repos.