aredn/aredn_ar71xx #16: Hardware Compatibility matrix

sarcasticadmin, 2018-06-18:
Would it be possible to mention the aredn.org hardware compatibility matrix in the `README.md`? https://www.aredn.org/content/supported-platform-matrix

That way users can easily lookup the hardware supported by this project.

ae6xe, 2018-06-18:
The README is actively evolving and I was thinking to capture directly in the README the list of devices, which image to use, and level of stability, keeping a record at a release or branch-build level.

sarcasticadmin, 2018-06-18:
@ae6xe that would be great! I would really appreciate a matrix with that level of detail. I am planning to purchase some new hardware for this project soon

ae6xe, 2018-06-22:
Added updates to README:

https://github.com/aredn/aredn_ar71xx/commit/62e95f140920cc304fa40196618161453fc2706e

closed 2018-06-22 by ae6xe

sarcasticadmin, 2018-06-25:
Awesome, thanks @ae6xe!