aredn/aredn_ar71xx #56: vtun package is missing

dman776, 2018-07-12:
build 36 does not contain vtun package

ae6xe, 2018-07-12:
This latest branch of openwrt does not by default build all the packages.  We need to explicit add a config entry for each package that needs to be built.  The issue is beyond vtun.   Current list of used add-on packages to include:
1) vtun
2) hamchat
3) tcpdump

What else?

dman776, 2018-07-12:
i need to lookup all the dependencies for vtun
remove hamchat
add meshchat
add iperf

Orv, 2018-07-12:
add iperfspeed!

ae6xe, 2018-07-12:
by adding a package to be built in the config file definition, e.g. vtun, the dependent packages will automatically be built and do not need separately added.   We just need top level package list to build.

meshchat is 3rd party contributed and distributed?

ae6xe, 2018-07-12:
iperfspeed is 3rd part contributed too, as I recall, but we can talk to the creator...

dman776, 2018-07-12:
yes.  I sent a note to Trevor asking if github was current.  https://github.com/tpaskett/meshchat and https://github.com/tpaskett/iperfspeed

ae6xe, 2018-07-12:
Need to investigate the options, this isn't an issue of how to include in the AREDN compile process to install the meshchat and iperfspeed packages.  It's a different problem to solve of how to reference and/or include these 3rd party compiled packages into the release (or how a live node can automatically refresh and find this package for download directly from the 3rd party).    Would help if we make this a different issue to track what to do on this.

dman776, 2018-07-12:
Agree.  Let’s make it a new issue (feature request)

Thanks,
Darryl

ae6xe, 2018-07-14:
vtun is now compiling in, but is not installed in default images per design with 7c163de .
tcpdump was already distributed in the image.
iperf package is already built. 

That completes the ideas we had on this issue, so I'll close this issue.

closed 2018-07-14 by ae6xe