aredn/aredn_ar71xx #57: Addition to sysinfo.json file

r1de, 2018-07-12:
Eric, KG6WXC here.

Could the nodes current time, node uptime and load average be added to the sysinfo.json information?

I'm sorry, I do not quite yet understand how to "fork" on github (or at all really).
Hopefully I will understand more about that soon! :smile: 

The needed Lua code might be something like this:
```lua
-- UPTIME CURRENTTIME AND LOADAVG
local uptname="/usr/bin/uptime"                                               
local curtime=""                                                         
local uptime=""                                               
local loadavg=""
if file_exists(uptname) then                                                  
        upthndl=io.popen(uptname)
        if upthndl~=nil then                                                  
                uptime=upthndl:read()                                         
                upthndl:close()      
                curtime=uptime:match("%d%d:%d%d:%d%d")
                loadavg=uptime:match("%d+%.%d%d,%s%d+%.%d%d,%s%d+%.%d%d")
                uptime=uptime:match("%d+%s%a+,%s%d+:%d%d")               
        end                                                              
end                                                                    
info["loadavg"]=loadavg                                                
info["node_current_time"]=curtime                                      
info["uptime"]=uptime                                                  
```

dman776, 2018-07-12:
Basically, the fork/pull request process is:

1.	Create an account on github
2.	Go to github.com/aredn/aredn_ar71xx
3.	In the upper right, click FORK
4.	Clone your fork into a local repo
a.	cd /yourdevdirectory
b.	git clone https://github.com/KG6WXC/aredn_ar71xx.git
c.	git branch yourbranchname
d.	cd aredn_ar71xx
e.	cd files/www/cgi-bin
f.	edit the sysinfo.json file (or whatever)
g.	git commit 
h.	git push origin yourbranchname
i.	in github on your repo:
i.	select the your new branch in the top left dropdown
ii.	click new pull request
iii.	describe your changes in the comments
iv.	click “Create pull request”

dman776, 2018-07-12:
even easier.  something like this....  (pulled from my https://github.com/aredn/aredn_ar71xx/pull/11 )
--------

mynix=nixio.sysinfo()
info['sysinfo']={}
upsecs=mynix['uptime']
info['sysinfo']['uptime']=secondsToClock(upsecs)
info['sysinfo']['loads']=mynix['loads']
for n,x in ipairs(info['sysinfo']['loads']) do
  info['sysinfo']['loads'][n]=round2(x,2)
end

ae6xe, 2018-07-12:
Clone your fork into a local repo
a.	cd /yourdevdirectory
b.	git clone https://github.com/r1de/aredn_ar71xx.git
**c.	cd aredn_ar71xx    <-  swap in order
d.     git branch yourbranchname   <- swap in order**
e.	cd files/www/cgi-bin
f.	edit the sysinfo.json file (or whatever)
g.	git commit
h.	git push origin yourbranchname
i.	in github on your repo:
i.	select the your new branch in the top left dropdown
ii.	click new pull request
iii.	describe your changes in the comments
iv.	click “Create pull request”

r1de, 2018-07-12:
Thanks for explaining guys! :smile:
I knew it was easy, I just get intimidated messing with others code sometimes.

I tried Darryl's example and I cannot get it to work, probably due to "secondsToClock" not being a function in sysinfo.json.

I can create the function, but to me, it seems simpler without it.

dman776, 2018-07-13:
Check my PR code. It’s in there

Thanks,
Darryl

r1de, 2018-07-13:
Please forgive me for sounding stupid... PR code?
Your Personal Repository Code?
I've been looking in there and I can't find that function anywhere.
Did you add it to here?
https://github.com/dman776/aredn_ar71xx/blob/develop/files/www/cgi-bin/sysinfo.json
I do not see it, I've also cloned your aredn repo and grep'd it and can't find any reference to "ToClock".

Also, I have to admit, this change has to do with my mapping system, I think that having the nodes current time displayed would be helpful to remind people to check that setting if needed.

r1de, 2018-07-14:
Done.

Took me a few tries to get it right but it looks good now.
Thanks for letting me learn. I hope to contribute more in the future! :smile:

dman776, 2018-07-14:
Sent you a comment to make a small change. 

To do so:

 

Make the change in your local branch.  

Commit to your local branch.  

Push to your personal github repo (“origin”)

 

(That should update the pull request)

r1de, 2018-07-14:
Ok. what should it be then? I don't want to mess up your guys versioning scheme.
1.2? 1.1.1?

r1de, 2018-07-14:
should close this since it is done....

closed 2018-07-14 by r1de