aredn/aredn_ar71xx #59: Make config option configurable at UI, and "sticky"

Orv, 2018-07-13:
For interfaces of type dtdlink, make olsrd config parameter "option Mode 'ether' " configurable at the UI, and "sticky" so it maintains across reboots, config changes, and firmware upgrades.

ae6xe, 2018-07-13:
This is the issue where you have manually changed or removed "option Mode 'ether'" on the dtdlink definition (because it is actually riding over a part 15 or other RF link).     If so, I'd recommend changing this request to state something to the effect of -- enable an out-of-box setting (which is inherently preserved across firmware upgrades). to enable dtdlink to be one of 2 options, A) cat5 cable; or B) rides over RF link.

Orv, 2018-07-13:
Works for me!

Orv, 2018-07-13:
Question - what about sites where some of the DtD links are RF and others are Ethernet?  There's no way to configure that unless you break it out on a per-link basis, right?  How the heck would you do that??

ae6xe, 2018-07-13:
One of the implementation options is to create another vlan tag for a
dtdlink over RF, say vlan 3.   the vlan switch would break out to cat5
dtdlinks vs RF dtdlinks, they'd be different virtual interfaces to the mesh
node and olsr with different config settings.

Orv, 2018-07-13:
That could work - another chunk of complexity added.  Is it worth it?  How much effort to implement?

cmlara, 2018-07-13:
Months of effort? 90% of this was worked out months ago already.

Originally the AREDN team agreed on a weekly call that this would use a section of of 172.16.x.y space just like tunnels used it would be up to admins to avoid collision. It would be up to each admin to configure it for their network.

(This is what I’ve had to deal with for years I would be on a call discussing something we plan how to inpment it and everyone forgets how to do it weeks later and we go back to square one)

It would be on vlan 3 

Zone routing rules was nothing, it would exist in the currrnt zone rules exactly the same as dtdlink doesn(you literally just add the new interface to the same rules as dtdlink if it’s enabled)

On a weekly call you guys had actually committed to add this to the backend as an advanced config option (officially supported but requiring cli usage) in 3.17.1.0 but never did and backed away from it.  It was scheduled to have a full GUI after the LUCI conversion. 

Quick to do if you understand the firmware and have developed good coding standards.

ae6xe, 2018-07-13:
Great, go ahead and make the changes and submit.

ae6xe, 2018-07-13:
Conrad, this is an inappropriate place to air your grievances.  Please do so elsewhere.   My name is not on any trademark or company and I'd like to focus on contributing to the community.

ae6xe, 2018-07-15:
Closing this thread to create a new issue with more clarity in the new feature and stay on topic.

closed 2018-07-15 by ae6xe