aredn/aredn_ar71xx #75: Firmware Upgrade issue

Orv, 2018-07-17:
Both Eric KG6WXC have observed that when upgraded to recently nightly builds, generally the upgrade will not take the first time; the node will reboot and wind up on the original version.
The second attempt is generally successful.
This is after the nodes have been running a couple of days.  Out of free RAM, maybe?

dman776, 2018-07-17:
Was this on a 64MB device?  If it was a 32Mb device, they are known to be unstable as noted in the README.md file

Orv, 2018-07-17:
I just attempted an upgrade from nightly build 55 to 59 on a Rocket M5 XW that had been up 23 1/2 hours.   It uploaded, went to the "upgrading firmware" page, and my browser lost connection with the node.  After several minutes, I was able to reconnect to the server.  It was still on version 55 AND the uptime still said 23 1/2 hours.
A second attempt was also unsuccessful - uploaded, said it was upgrading, lost connection, regained connection, still on version 55, still 23 1/2 hours uptime.
I have a Support Data file for this, but "We don't support that file type."   I just renamed it to .zip to upload it.  You'll have to rename it back to .tgz.

[supportdata-WD6EBY-LA-MtWilson-SW-Sector-5G-201807170833.zip](https://github.com/aredn/aredn_ar71xx/files/2202343/supportdata-WD6EBY-LA-MtWilson-SW-Sector-5G-201807170833.zip)

Orv, 2018-07-17:
Note - it's possible that network issues may be preventing a complete upload.  We'll do some more research.

Orv, 2018-07-17:
Darryl, let's close this issue for now.  More research is required, as two of three presumably problematic nodes took the 59 nightly build first time.    We'll focus on troubleshooting the third.

jbrady42, 2018-07-17:
I had this happen during testing of nightly builds. First upload resulted in reboot but no change. After a second upload the firmware took. I have not seen any cases other than the first, but will drop the support data in here if it happens again.

dman776, 2018-07-17:
ok.  closing this as "invalid" for now.

closed 2018-07-17 by dman776