aredn/aredn_ar71xx #81: Completely remove IPv6 to save resources

dman776, 2018-07-19:


dman776, 2018-07-25:
IPV6
SNMP
others?

Orv, 2018-07-25:
Not SNMP please.  We use it heavily to monitor our network performance.  (Pulling it from the 32MB image would be OK, just not the 64 MB).

r1de, 2018-07-26:
That works for me.
what about removing the usb support is it actually needed?
kmod-usb-uhci
kmod-usb-ohci
kmod-usb-core

and what about these 2:
odhcpd-ipv6only
libip6tc

Orv, 2018-07-26:
SNMP as an option would be OK.  We'd mostly want to monitor high-level nodes and they'll all be 64 MB devices.

r1de, 2018-07-27:
I have been experimenting. :smile:

**libip6tc** should not be removed yet, iptables breaks when it is not available. Oddness occurs.

**kmod-usb*** I made into modules, there are 5 of em.

**tcpdump-mini** and **libpcap** can both be made into modules.
Most "normal" users are never going to need them, and those of us who do can install it if/when needed.

**smnpd** and **libnetsnmp** I also made into modules just to see.

With all those removed the resulting image is about 5.7MB, when loaded I had 1.5MB left in my "flash" area on an old RocketM2.
Even after installing tunnels there is still about 700K left free in "flash"!

Everything is working fine. Even the tunnels.

I am sure there are a few more packages that this can be done too...
Just don't try to remove libip6tc!!
The line about libip6tc should be removed from common.config just in case!

r1de, 2018-07-28:
I can't stop messing with this...
I am pretty sure I removed *all* ipv6 now.
Image file down only a little bit this time, to 5.6M.

I was able to build an image and load it onto an old NSM2 I have, and it is working just fine it seems so far. (Even on the large So. Cal. network)
The NSM2 still has about 5.5MB free in Memory and about 1.6MB free in Flash.

I'm not sure this should be called "done" until there is more testing tho.

dman776, 2018-07-28:
submit the PR and we can start reviewing it

r1de, 2018-07-28:
I will, but now that I said something, I am double checking some things first. :smile:

Orv, 2018-07-29:
I loaded Eric's firmware version on an old 32 MB NSM2 and it seems to be running well (I found one possible bug unrelated to his work that we'll open an issue on.
So far so good...

commit 52a8b79f (2018-08-02): Trim Packages (#96)

Closes #81 

* Trim Packages

Kernel support for USB has been moved to modules
libpcap and tcpdump have been moved to modules
libnetsnmp and snmpd have been moved to modules
IPv6 has been removed from the kernel
IPv6 support has been removed from busybox network utils as well.
(no more funny DNS responses in nslookup!)

* Remove ping6 and traceroute6

They are no longer needed

* Changed patch name and number and updated series file
closed 2018-08-02 by dman776