aredn/aredn_ar71xx #85: nslookup error

Orv, 2018-07-20:
On Nightly Build 59, nslookup works, but throws this error, which may be causing other issues we've seen (not ready to talk about those yet):

root@WD6EBY-LA-MtWilson-SW-Sector-5G:~# nslookup w6bi-vc-qth-2g
Server:         127.0.0.1
Address:        127.0.0.1#53

Name:      w6bi-vc-qth-2g
Address 1: 10.62.29.243
***** Can't find w6bi-vc-qth-2g: No answer**
root@WD6EBY-LA-MtWilson-SW-Sector-5G:~#

ae6xe, 2018-07-20:
The code for the operation of the node is using nslookup to do the reverse ip-to-hostname lookup.  (At least that's all I found when searching all the code.)   While you might get this from the command line, hostname-to-ip, the output wouldn't affect the node operations since it's not used.  When ready, post any failures observed.

Orv, 2018-07-24:
Keep this open for now; Doug, WB2YXY, who uses bind and dnsmasq together, reports that the bind response back from the nightly builds differs from the (correct) response back from the 3.16 production release.  Hope to have more coherent information soon.

Orv, 2018-07-25:
We can close this; we won't be pursuing it for now.

closed 2018-07-25 by dman776