aredn/aredn_ar71xx #93: Firmware update screen missing

ai6bx, 2018-07-25:
I updated two nodes tonight and note that neither went to the updating splash screen and rather just stayed on the admin page with a footer stating awaiting my node. After a few minutes I finally was bold and hit the refresh button and found the update had actually taken place. It would be helpful to have the splash screen back indicating things are happening.

Btw, so far the nodes, a Rocket M5 XW and a pbe 400 iso, are working well.

ae6xe, 2018-07-26:
I've also noted this is intermittent, sometimes shows, sometimes does not.   Also, when it does display, with a notice of reconnecting after ~3 min, it jumps to a disconnected page before this point.   (I speculate the openwrt sysupgrade command is further stopping processes to free up resources, and this may be a casualty.)

ai6bx, 2018-07-26:
Thank you,

I am assuming then as long as the node does come back after refresh and shows the new firmware all is good.

k1ky, 2018-08-24:
For "best" results, suggest that you start from a freshly rebooted node before performing any upgrades.  The full message "usually" displays when following that procedure.  I found that nodes that had been running for long periods, days, weeks, months seem to be the ones that don't show any or all of the text during upgrade.

ae6xe, 2018-08-24:
I've not been seeing this symptom recently with recently booted nodes.  The other changes to reduce RAM usage may have contributed to resolving.   Please post if seeing (or not seeing) these symptoms (and conditions/environment if the symptoms are still occurring).

Orv, 2018-08-26:
While installing build 200 on a dozen and a half nodes today, I no longer see this issue.  My two cents.

ai6bx, 2018-08-27:
Sounds like it is time to load the next nightly build. Does 200 address the gigaport conflict in the PowerBeams?

dman776, 2018-08-29:
this issue has reported to have been resolved.
(AI6BX, can you open a new issue on the PBE Gigabit issue?)

closed 2018-08-29 by dman776