aredn/aredn_ar71xx #94: TP-Link CPE210v2 and v3

ZL2ARN, 2018-07-26:
Thank you for inviting me to this forum.
Thank you Andrew for the excellent work on the CPE210v3.
Both v2 and v3 updated to the latest snapshots.
When the "Mesh Gateway" is not active, I have NO access to the Internet.
When the "Mesh Gateway" is active, I can surf the Internet from the WIFI network.
"Upload data to the AREDN Server" appears to work as the node reports "Server updated" after some 30 seconds. However AREDN maps do not show this update. Had a look around the World just in case the updated data was corrupted but could not find my Node anywhere.
Tunnel Server/Client will not install, and get an error message. I understand Joe may have some thoughts about this.
The Ubnt nano m XW was updated and all of these functions are working nicely.

dman776, 2018-07-27:
FYI.. there are 2 nodes in the database:  ZL2ARN-TUN-CLIENT" and ZL2ARN-TPLink-CPE210v2"

ZL2ARN, 2018-07-27:
The only two of my nodes I can find in the map2 data base is ZL2ARN-TUN-CLIENT, and it does NOT appear on the AREDN map, it is a Ubiquiti node, and the other one is ZL2TB-NSM2-XW, a Ubiquiti node and this one does appear on the AREDN map.
I cannot see any of my TP-Link CPE210 nodes in the map2 database or on the AREDN map.
My current test nodes are:-
ZL2DRN-Test-v3
ZL2ARN-TPLink-CPE210v2.local.mesh |  
ZL2DRN-TPLink-CPE213v3.local.mesh
ZL2TB-TPLink-CPE210v3.local.mesh
Each of these nodes reported "successful update to the AREDN server"
I am now going to make the Lat/Long different for each device, just in case the server does not like more than one node with the same co-ordinates.

ZL2ARN, 2018-07-30:
Busy looking through and understanding the 250 odd lines reported with "wireshark". Comparing the NSM2-XW which is performing ok, against the CPE210v2/3. Nothing obvious so far. We do have access to the Internet and the DNS lookups appear to be reporting ok.

ZL2ARN, 2018-07-31:
Think I may have found the problem with the TP-Links.
Wireshark is showing lines in RED, an IP of 23.185.0.4 which does not appear in any of the DNS lookups, and the "whois" lookup suggests to me that it is not associated with AREDN. This IP does not appear in the Ubnt NSM2-XW Wireshark table.
23.185.0.4	TLSv1.2	114	Application Data
23.185.0.4	TLSv1.2	99	Encrypted Alert
And then the lines following this are in RED

Note.
Ubnt NSM2-XW does not have any "Encrypted Alerts"

ZL2ARN, 2018-08-02:
Ok guys, someone has made some changes and it was not me.
The AREDNMesh  map updates are now working on the TP-Link CPE210v2 & v3 !!
The updates were working before I updated to the latest snapshots, and still working nicely after doing the "sysupdates".
I have all my test nodes nicely spaced out on the AREDN map.
So I guess this ISSUE can now be closed.

dman776, 2018-08-02:
Yes.  I made changes to the Kml generation script and committed those a few days ago.  I added the new mikrotik and tplink models.

Thanks,
Darryl

closed 2018-08-02 by dman776