aredn/aredn_ar71xx #100: Add support for Mikrotik hAP AC Lite

dman776, 2018-08-01:
add support for the mikrotik hAP AC Lite to be a qualified replacement for UBNT AirRouter
this device is dual band (2Ghz/5Ghz) and could be a node and access point at the same time

ae6xe, 2018-08-02:
The specific model is RB952Ui-5ac2nD.    This is a 64Mb RAM device.   There are some hAP lite models with only 32b RAM and not looking at this models for now.

K9CQB, 2018-08-23:
This is great that someone is working on this device. I just bought a RB952Ui-5ac2nD for $49 and plan on getting a few more for our links out here in northern Virginia/DC. 
Thank you, 
-K9CQB

ae6xe, 2018-08-23:
This is to the point where AREDN firmware is loading on this device, the ports are all configured for LAN, WAN, DtDLink.    Starting now to work on the code to handle 2 wireless interfaces to configure.   The first availability in the nightly builds will be mesh RF links on 2Ghz and a mesh AP on 5Gz (802.11ac).

ae6xe, 2018-09-06:
https://github.com/aredn/aredn_ar71xx/commit/0bd057ed40bb26cf9b321ad725a441cf689eb6ac