aredn/aredn_ar71xx #110: AREDNMESH-Live maps

ZL2ARN, 2018-08-04:
There is a large number of redundant Nodes appearing on the Google map.
It is nice to know who has their Node  QRV, but it is also nice to know who is not there any more.
For instance, my test nodes will soon be QRT. And I think they should then be removed from the Google map. So how to achieve this ??
How about a monthly update from each node that has the Lat/Long set in the node.
The we could do a monthly purge on the AREDNMESH google map server.
This would keep the map really up to date with real time info.

dman776, 2018-08-04:
can you provide an example of a node name that is redundant on the map?

ZL2ARN, 2018-08-06:
Darryl,  VK4YOI-5-NSM2 is a good example to confirm redundancy. I have E-mailed him to confirm his status, but so far no response.  His SSID suggests a local WIFI network.
ZL1AXG-AG-QTH-WLG I can confirm is QRV but on our Wellington SSID Network so not visible on AREDNMEsh Internet Network. His QTH is not in the middle of the Tasman Sea, and I have asked him to correct it.
All of my TP-Link test nodes will return to the Wellington WIFI network  after we have completed our "snapshot" testing, and so they too will become redundant.
I suspect there are many WIFI networks up your way. So do we need to see WIFI networks that are not linked to the AREDNMesh network appearing on the World map ??
I will extend my search to confirm other redundant nodes.

dman776, 2018-08-12:
I'm only seeing on VK4YOI-5-NSM2 in the KML files (viewed with Google maps).
[image]

dman776, 2018-08-12:
Same on the online web map...
[image]

ZL2ARN, 2018-08-13:
No response from my E-mail to VK4YOI, so I cannot identify his current status.
My apologies for starting this "low" priority issue. We have much more serious firmware issues with  higher priorities to resolve. 
Perhaps we can get back to this issue at some later date.

closed 2018-08-13 by dman776