aredn/aredn_ar71xx #127: Generic Linux Node Daemon

blm2008, 2018-08-09:
This is a feature request carried over from a forum post [here](https://www.arednmesh.org/content/tunnel-server-using-raspberry-pi-3b).

It is explained in the above forum discussion that any member of the mesh, including tunnel servers, need to have OLSR and other node software.  [Current guidance](https://www.arednmesh.org/content/32-mb-nightly-build-now-available) is also that "none or one tunnel is all you should run on a 32 MB device."  [Other recent discussion](https://www.arednmesh.org/content/software-updates-and-tunnel-software) has indicated that tunnels are meant to be over the internet and not a private (non-internet) LAN.  

Therefore, I would like to install a tunnel server daemon on an external Linux server that is (1) not memory-limited and (2) connected full time to the internet with a static IP address.  Each of my 32 MB Ubiquiti-based nodes would connect to that tunnel host and thus would only need to manage one tunnel client as suggested above.  This could be simplified by not needing a web GUI or node status pages, etc, only a config file to list allowed nodes and their passwords.

Could a linux package or source code be made available to install OLSR and other requisite software on a generic host for the purposes of hosting a tunnel server?

ae6xe, 2018-08-09:
My thoughts on this, not sure if this reflects others thinking or not...

I'm a bit concerned from a practical support perspective in creating an install for a general computer.  There can be lots of issues with OS versions, linux flavors, etc. that make it problematic to get installed and maintain.  This consumes time not working on core features for AREDN mesh nodes.   From this perspective, this might be a project for someone other than those of us focusing on the mesh nodes.   

I'd think the best way to do this on special purpose device is to package it up as an 'appliance'.  This might be a pre-made image for a given hardware platform, e.g. rasbpi, etc.  It's a matter of known-how to make happen, but doable.

Let me propose an alternative option to consider.    We are working on an AirRouter upgrade to a 64Mb RAM device, the Mikrotik hAP ac lite RB952Ui-5AC2ND model.   It has WAN port, 3 LAN ports, 1 DtDLink port, and 2 wireless adapters (2Ghz and 5Ghz).   We expect there to be a mesh node RF link as well as a mesh LAN Access Point for the shack.  This is an all-in-one device for only ~$45.  

A rasbPi and other low cost computers like this won't scale well to the 40+ tunnel connections.   However, you could take this device and run the 10 max tunnel connections on it and just get 4 of them to have 40 tunnels.   It spreads out the point of failures and is still lower cost than purchasing a general purpose desktop/laptop computer.   the same thing could be done now with  any of the 64Mb RAM node devices, but packaging for a tower is not ideal and vlan switch is needed.

blm2008, 2018-08-10:
Thanks for the comments on this, guys.  I think Joe's suggestion of using the Mikrotik hAP when the firmware is ready will be a good option for most as an AirRouter replacement. 

We'd be interested in a stand-alone Docker container when time/priorities allow, or if someone would like to take this on as a sub-project.  It would be a more flexible and less hardware-dependent solution.

closed 2018-12-04 by ae6xe