aredn/aredn_ar71xx #138: Nanostation M5 Tunnels

n3bkv, 2018-08-13:
Similar to the previously mentioned issue with TP-Link I am also unable to install the tunnel server on an NSM5 XW - Error message reports: installation failed. I am also seeing strange networking issues through the box where Internet web pages take a much longer time to load.

ae6xe, 2018-08-13:
Can you attached a support download of the mesh node?    The NSM5 XW would need a cat5 cable on the secondary port, then plugged into a vlan capable switch configured to have access to the internet.

n3bkv, 2018-08-13:
Hi Joe,
Attached is the support download.  
Yes - the node has a secondary cable installed going into a vlan capable switch.  Everything functions correctly on version 3.16.1.1.
Dave

dman776, 2018-08-15:
I just successfully loaded nightly #165 on a Nanostation M5 XW, then, clicked the Install Tunnels button.  No issues here.

ae6xe, 2018-08-15:
I did the same on a NSM3 XM and it succeeded.     Maybe I'm not finding in github issues, but is there an attached support file?

dman776, 2018-08-15:
To attach files, you must drag them into the comments box (you cannot attach them to a reply email).

r1de, 2018-08-16:
I'll let him know the report file never made it.
He reverted to 3.16.1.1 and his setup went back to normal.

r1de, 2018-08-16:
I had n3bkv forward me his support file from when he was having issues.
Unfortunately, I cannot seem to attach it here, I just keep getting, "...can't process that file".
I tried renaming it and a few other things, nothing seems to work.

Please find the file here:
https://kg6wxc.net/supportdata-N3BKV-LA-Burbank-5G-201808130653.tgz

n3bkv, 2018-08-17:
> The WAN interface shows that it does not have an IP address and is not connected to the home network to access the internet and download the tunnel packages.
>
> eth0.1 Link encap:Ethernet HWaddr 80:2A:A8:F3:66:B6
> UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1
> RX packets:0 errors:0 dropped:0 overruns:0 frame:0
> TX packets:110 errors:0 dropped:0 overruns:0 carrier:0
> collisions:0 txqueuelen:1000
> RX bytes:0 (0.0 B) TX bytes:37620 (36.7 KiB)

Hi Joe,
Thank you for the info below - I found the issue.  My Internet VLAN had some strange routing going on that 3.16.1 allowed somehow.  
I have resolved the issue and I can now download the tunnel software on the node and everything is functioning correctly.
Feel free to close the ticket.
Thank you for all of your help.
Dave

dman776, 2018-08-17:
Thank you for the info below - I found the issue.  My Internet VLAN had some strange routing going on that 3.16.1 allowed somehow.  
I have resolved the issue and I can now download the tunnel software on the node and everything is functioning correctly.
Feel free to close the ticket.
Thank you for all of your help.

closed 2018-08-17 by dman776