aredn/aredn_ar71xx #141: Add support for Powerbridge m5

txcoder, 2018-08-19:
Our wisp recently decommissioned a number of Powerbridge M5's that I'd like to integrate into our local mesh. I've attached dmesg and device.info files and am willing to test firmware. I may be able to help with the development if someone can give a quick primer on supporting new hardware.

[board.info.txt](https://github.com/aredn/aredn_ar71xx/files/2300546/board.info.txt)
[dmesg.txt](https://github.com/aredn/aredn_ar71xx/files/2300547/dmesg.txt)

ae6xe, 2018-08-19:
the dmesg shows this device has the same Ethernet port chip AG7240 (eth0 & eth1),  wireless chip AR928x, and flash chip AR724-nor0  as a NanoStation M5 XM.   If it were me, I'd first try to load the nightly build "nano-m" factory image (not the "nano-m-xw").      If lucky, it will come up withe AREDN image, but would need one entry in /www/cgi-bin/perlfunc.pm to add the board id and power settings. 

It would come up in first boot and in AREDN basic setup, the selection of channels would be incorrect/incomplete.      I'd expect the nightly builds will load via tftp, and maybe from the AirOS UI depending on the bootloader version.  You would not need to downgrade AirOS bootloader with the nightly build.

dman776, 2018-08-19:
is there one or two ethernet ports on it?   I think OpenWrt was suggesting the rocket-m-xw firmware...

ae6xe, 2018-08-19:
dmesg shows eth0 and eth1 identical to NS M5 XM.   the secondary port eth1 is not configured for any device in this configuration in AREDN and prior firmware versions.    The boadid shows an AirOS prompt of "XM.v5.6.3#"

txcoder, 2018-08-20:
It has a single ethernet port. 

I'll try loading the nano-m image this week as time allows - will the /www/cgi-bin/perlfunc.pm need the info contained in the device.info file?

ae6xe, 2018-08-20:
If the Ethernet port is not functional after loading the nano-m image (http://192.168.1.1:8080 does not respond), then look for an AP with SSID of "meshnode" to connect to at the same address.   Sometimes, an image doesn't have the driver for the Ethernet, but the wireless is working.  The basic difference in images is:

Rocket = Bullet + USB
Nanostation = Bullet + additional ethernet

I do see a USB driver in AirOS.  Is there also a USB port on the device?  AREDN firmware isn't configured to use this, but that would suggest using the Rocket image.    What is important is having a functional ethernet port first. 

If nano-m isn't working, try the rocket-m image 2nd, then the bullet-m in that order to see which one works.   Inspect the other entries in perlfunc.pm.   there is a setting for offset and max power.  As I recall, take ubiquiti board info max power minus offset to enter the max power in perlfunc.pm entry. 

the internal 2nd port is not used or physically extended, if only one port on the packaging.  But we can see it in the AirOS dmesg:

[    8.552000] AG7240: fifo cfg 3 01f00140
[    8.552000] AG7240CHH: Mac address for unit 0
[    8.552000] AG7240CHH: 00:27:22:97:4e:3d
[    8.758000] AG7240CHH: Mac address for unit 1
[    8.758000] AG7240CHH: 02:27:22:97:4e:3d
[   18.444000] device eth0 entered promiscuous mode
[   18.456000] device eth1 entered promiscuous mode

txcoder, 2018-08-21:
The powerbridge successfully accepted the nano-m image. So far it is connected to a nanostation m5 without any trouble. I added the following to the perlfuc.pm file:

`    '0xe1a5'=> {
      'name'            => 'PowerBridge M5',
      'comment'         => 'PowerBridge M5 (in testing)',
      'supported'       => '-2',
      'maxpower'        => '22',
      'pwroffset'       => '5',
      'usechains'       => 1,
      'rfband'          => '5800ubntus',
     }
`

ae6xe, 2018-08-22:
good news.   Would you like to do a github pull request of this change and results to make official and get your callsign/name on the contributors list?   Can you attach a support data download from the Admin page, link at bottom.   I'd like to inspect, to confirm all looks good.

dman776, 2018-08-22:
Does DTDLinking and WAN access work properly?

txcoder, 2018-08-22:
I'll test DTD and WAN tomorrow then submit a pull request when we verify its stable . Support data is attached.
[supportdata-KG5REO-PB5-PRC-201807301911.tgz.zip](https://github.com/aredn/aredn_ar71xx/files/2308613/supportdata-KG5REO-PB5-PRC-201807301911.tgz.zip)

txcoder, 2018-08-22:
#147

dman776, 2018-08-22:
so DTD and WAN access tested ok?

txcoder, 2018-08-22:
Yes - it created a DTD connection with other nodes on vlan2 and pulled an IP via DHCP on vlan1. After that it performed appropriately as a WAN gateway.

ae6xe, 2018-08-22:
commit to repo:  8d3b63a

closed 2018-08-22 by ae6xe

ae6xe, 2018-08-23:
txcoder, what is your callsign to add to contributor's list?  Please add to this file at top level directory and submit another PR?

ae6xe, 2018-08-23:
The top level README.md also needs a PR to add this new device.