aredn/aredn_ar71xx #144: TP-Link CPE210v2 & v3 Tunnel-Linking

ZL2ARN, 2018-08-21:
After messing around now for a few days, I have had no success getting an active tunnel (Server/Client) to activate and link. 
Checking through all the gui Client and Server entries, all seems in order, except the Tunnel Server "client" field reverts to upper case text when attempting to save any changes. Is this significant ?
All other fields in both Client and Server pages can be either upper or lower case and remain so when saved.
These two screen shots are from " :/etc/config.mesh/vtun" and all appears to be in order. !!
And showing the upper case that refuses to be set as lower case.
[image]
[image]

dman776, 2018-08-23:
the uppercase is intentional.   
it does appear that your server is listening for connections (I tested from here and get the proper VTUN response).
Check your settings on the server side node to ensure the matching password and connection name in vtundsrv

ZL2ARN, 2018-08-23:
That was the old bullet 3.16.1.1.node you most likely saw. It is now QRT while I play with the CPE210v2
I now have the CPE210v2 in its place with exact same settings and via the vlan switch.
I can confirm the Tunnel server is now linking to the remote clients. 
However the one and only local tunnel client I have is not linking to its remote server.
Working on it !!

ZL2ARN, 2018-08-25:
Ooooops, sorry guys, the tunnel server is NOT linking to the remote clients. I was looking at the wrong screen. It was a long hard day.
I think I may now have found the problem. Comparing command line "ifconfig" network options on the operational Bullet 3.16.1.1 and the Ubnt nano m2, I see that the "tun 60-00" is missing from the CPE210 v2 and v3 nodes. Perhaps this tunnel has failed to compile and is the reason for no tunnel linking.

dman776, 2018-08-25:
Did you install the tunnels with the “install tunnels” button, or, manually by loading packages?

Thanks,
Darryl

ZL2ARN, 2018-08-25:
Hi Darryl,
It was the "install tunnels" button on the "tunnel client " page, but I do believe I have used also the "install tunnels" button on the "tunnel server" page on other occasions. They have all installed nicely with no problems using these buttons.
Oh, I see we are up to AREDN-192 Snapshots, I had better catch up.

Thanks
Gordon

ZL2ARN, 2018-08-25:
Updated to latest aredn-192-5869 (sysupgrade) but same results, no "tun 60-00" , still missing in both cpev2 and cpev3.

ZL2ARN, 2018-08-26:
Running services (ps command) shows:-
vtund[c} : ZL2ARN-TUN-CLIENT........tun tun60  for 3.16.1.1 firmware in bullet (active link)
vtund[c] : ZL2ARN-LINK NORTH........tun tun60  for Ubunt nano m2 (active link using port 5526)
However the TP-Link cpe210cv and v3 have no "vtund" entry in this service list when the client/server are attempting to connect.

ZL2ARN, 2018-08-26:
Yes they **are** all there. I had to log-in manually as the DHCPd is **not working on the LAN port**. The DHCPd does not appear as one of the running services, and the usual LAN devices have no reservations from the Node. Yes we are working through the VLAN switch. The server tunnels came up very slowly, some 2-3 minutes, and the client tunnel has not yet come up. ! It does appear in the (ps) list as "running" but no assigned  tun number.

ZL2ARN, 2018-08-26:
Oooops...again. Updated to snapshot 200-41d... and now the DHCP is working nicely. Server tunnels are coming up quickly to,  only the client tunnel is still refusing to link. Checking once again the parameters for this client link.

ZL2ARN, 2018-08-29:
My CPE210v2 is on the Internet MESH, "ZL2ARN-TUN-CPE210v2", checking to see how it handles the large volume of OLSR traffic. Tunnel servers are working nicely, and still searching everywhere for why the Tunnel Client will  **not**  link to the far end. ??

dman776, 2018-08-29:
Will test with ae6xe this week with a cpe210v2 client

ZL2ARN, 2018-09-02:
Looking through "ifconfig" entries, I see that eth0.2 has a strange assigned IP. It is 10.49.65.96 ??
The WLAN0 assigned IP is 10.48.65.96 for the same Node. 
eth0.2 has a good number of TX bytes and is increasing with time. however eth0.2 has **zero** RX bytes. No activity on the RX port. ??
Perhaps this is the problem with the tunnel client.
The tunnel client was enabled but not active for this test. And it shows up under running processors as "connecting".

ZL2ARN, 2018-09-20:
Darryl,
I think we should close this issue. I am not having much luck confirming the Tunnel settings at the far end of my tunnel. I have restored my original Ubnt bullet that was linking ok and now that too is NOT linking any more. 
I will now find another link partner and continue with this tunnel  "client to Server" testing.
So in the interim, please close this issue.

closed 2018-09-20 by dman776