aredn/aredn_ar71xx #149: boardid: 0xe845 !!!! UNSUPPORTED DEVICE !!!!

mathisono, 2018-08-23:
Hoping to get rid of this if i upgrade to the new RC

!!!! UNSUPPORTED DEVICE !!!!
boardid: 0xe845
Device has not been tested. Please file a ticket with your experiences.

dman776, 2018-08-23:
3.16.2.0RC1 does not add any new hardware support.   Check the nightly builds for that

ae6xe, 2018-08-23:
Thanks.  There is a lot of history and information on the loco M5 XW.  This banner only says unsupported because most all the XW devices have not been included in any prior release.  However, this device is considered stable in the nightly builds and people have had this device on the mesh live for over a year with no issues.   We'll remove this statement on all devices very soon when the nightly builds turn into a release.    Please upgrade all XW devices to the current nightly build, it is the most stable option and in 2 or 3 months expected to become a release.

mathisono, 2018-08-23:
thanks! Currently its more stable than the NS-coco m2. tho I think its hardware failure in that case....