aredn/aredn_ar71xx #160: Feature Enhancement: Show OLSR Routing Table Size

n2mh, 2018-08-27:
Enable a new type of statistic to indicate the number of entries in the OLSR routing table. This statistic should be displayed on a node's main page as well as the Mesh Status page. Currently, the Mesh Status page (aka the Nodes List) shows most of this information but makes no attempt to quantify it. 
In addition, this statistic would count additional entries that are not nodes, reserved addresses or advertised services such as tunnel networks.

On many of my nodes I have an entry that reads "OLSR entries:" to display this number. I get this number by doing

ip routes list table 30 | wc -l

This essentially counts the number of lines in the OLSR routing table and returns a number which displays on the web page. In /www/cgi-bin/status, the code looks like this:

$nodecount = "<th align=\"right\">olsr entries</th>\n";
$nodecount .="<td>".`ip route list table 30 | wc -l`."</td>\n";

push @col2, $nodecount;

Visually, this stat appears right after memory usage.

By adding this new statistic, network operators may be able to get a handle on when a network grows too large for either available memory or cpu power to process a large routing table in a timely manner when making individual packet routing decisions.

Orv, 2018-08-28:
What's too large?  What's the criteria?

n2mh, 2018-08-28:
Well, that's the point... When the network starts to fall over, how many nodes does it take? Without a way to measure the size of the table, you won't be able to get a handle on that number.

We don't know what that magic number is. But, there is one and you will need to know what that is and know when to take steps to get the network manageable again. And, be able to compare with other groups as they grow.

The other use is to watch the size of the network. It it abruptly gets bigger or smaller, then something happened that should be looked into. If all of a sudden, you lost 100 entries, more than likely some backbone node died. If it grows by a couple of hundred entries, how did that happen? Is there some freak of propagation in progress? Is there something else going on?

dman776, 2018-08-29:
we need to do a memory profile for the 'ip route list' command since it will be consuming extra memory on the mesh status page (which is unprotected)

ae6xe, 2018-08-29:
the "ip" command is part of busybox.   Since this is used by many/most things, it would already be in RAM and not further duplicated.  The unique stack and data add-on memory for this process, should be negligible or minor.   Although the pennies add up over time.

r1de, 2018-08-29:
I changed my testing NSM2 to have this:
[image]

It's running "ip route" twice to get that info and the memory usage is staying about what I would normally expect to see on our fairly large network...

ae6xe, 2018-08-29:
looks good to me.

r1de, 2018-08-29:
PR #163 updated

PR #163 (2018-09-01): aredn: node description, lat/lon display, olsr entries

Added node description to mesh status page
Fixes #161

Added Lat/Lon (if available) to status and mesh pages.
It is displayed above the description and under the node name.
Fixes #145

Added OLSR Routing tables entries to status page just under the memory area
Also added this same info just above the buttons on the mesh status page
Fixes #160
commit ef6d092b (2018-09-01): aredn: node description, lat/lon display, olsr entries (#163)

* aredn: node description on mesh page, lat/lon on both status and mesh, olsr entries

Added node description to mesh status page
Fixes #161

Added Lat/Lon (if available) to status and mesh pages.
It is diplayed above the description and under the node name.
Fixes #145

Added OLSR Routing tables entries to status page just under the memory area
Also added this same info just above the buttons on the mesh status page
Fixes #160

* aredn: OLSR routing table entries

Small addition to the displayed info

* aredn: corrections

move lat [... cut]
closed 2018-09-01 by dman776