aredn/aredn_ar71xx #165: PowerBeam M5 400 network errors

KD6EPQ, 2018-08-29:
With certain switch and laptop combinations a PowerBeam M5 400 (only the non-ISO model was tested) exhibits network errors (FCS and fragments).  Setting both the switch and PowerBeam to 100/Full no autonegotation temporarily fixes the problem.

ToughSwitch 5 and MikroTik HEX PoE bring out the issue.  An HP 1810 gigabit switch, and Cisco 10/100 only switches did not show the problem.

A Dell and a Toughbook laptop, containing the Intel i219 and i218 NICs, and running Windows 7 both exhibited issues.  An older i7 Dell laptop, Win 7, with an older 85xxx NIC did not.

Problem exists as of nightly 206.

ae6xe, 2018-08-29:
Symptoms can also include node dropping off mesh periodically.  Root cause is related to devices with Gb ports and unable to reliably sustain this rate (likely exacerbated with longer cat5 cables).   
Workaround in use (install the ethtool package):

ethtool eth0
ethtool -s eth0 speed 100 duplex full autoneg off
put it in /etc/rc.local to persist

KD6EPQ, 2018-08-29:
Newer nightly builds use:
opkg install http://downloads.openwrt.org/releases/packages-18.06/mips_24kc/base/ethtool_4.15-1_mips_24kc.ipk

Older stuff use:
http://downloads.openwrt.org/chaos_calmer/15.05/ar71xx/generic/packages/packages/ethtool_3.18-1_ar71xx.ipk

ae6xe, 2018-08-30:
KD6EPQ,   I suspect this may be a root cause fix.  I could create an image to test out.  Up for a test?
  
https://github.com/openwrt/openwrt/commit/4e39e213af7e3e0cd747403e8c227e145cfef988

PBE-M5-400 XW: ag71xx ag71xx.0: connected to PHY at ag71xx-mdio.0:04  [uid=004dd072, driver=Atheros 8035 ethernet]

Locking the port to 100Mb works and may be sufficient for the throughput we have, but if we can fix the negotiation process, then it wouldn't limit when 1Gb speed is stable.  This would be a better solution.

KD6EPQ, 2018-08-30:
Let's give it a go.

ae6xe, 2018-08-30:
Here are test images.  This link will not be live after the images are no longer relevant. 

https://drive.google.com/drive/folders/1BN4sX8pA2CSxkhOGGZ0wbQOWYBdJ_5dh?usp=sharing

KD6EPQ, 2018-08-31:
Bug still persists.  Saw a slightly lower rate of FCS errors but the link is even slower ~200kbps and drops packets via iperf and pings.  The above fix still works and returned rates to 85+Mbps error free.  Tested on MikroTik gigabit switch.
I think we're on the right track tho... this is something in the NIC driver within OpenWRT.

ai6bx, 2018-09-01:
The issue began with 3.17 RC and continued with AE6XE201819. I have seen this both on the non-iso and iso versions. I have corrected, or minimized, with ethtool locking the node to 100/full and locking the switch to 100/full. I have both iso and non-iso I can test on.

ae6xe, 2019-02-23:
AI6BX reports that since the upgrade to nightly build 692-bb818a9, which included the Openwrt 18.06.2 point release.   These symptoms have not yet occurred, the work around was not applied.  Please monitor your nodes and if able, do not apply the work around to confirm this is resolved.

ai6bx, 2019-02-23:
I have it successfully running on two PowerBeam M5 400s at this point with no issues. I am continuing to monitor and have been 13 days without issue at this time.

 

Keith – AI6BX

AJ6GZ, 2019-03-06:
Still seeing this issue under nightly 746.  Tested against the usual MikroTik hEX POE.  There were noticeably fewer dropped packets  and FCS errors on the link but the throughput via iperf3 without the work around was terrible (1-5Mbps).  After the work around back to 70Mbps+.

Newest ethtool that works at this point is: http://downloads.openwrt.org/releases/18.06.2/packages/mips_24kc/base/ethtool_4.19-1_mips_24kc.ipk

dman776, 2019-03-26:
Can you re-test this with 3.19.3.0?

ai6bx, 2019-03-26:
I will load on two units tonight and monitor.

AJ6GZ, 2019-03-26:
3.19.3.0:

When set to gigabit/autoneg I am not seeing the FCS errors anymore!  However performance is still limited from iperf3:
 windows as server  ~35Mbps
 node as server ~75Mbps

With both sides locked to 100/Full, no autoneg:
 windows as server ~90Mbps
 node as server ~80Mbps

Both done with a large packet ping going in the background to add a little something extra.

Tested on MikroTik HEX PoE.  I can test against the Dell laptop (i219 chip) tomorrow if needed.

ae6xe, 2019-03-27:
This looks like a success.   I'm not aware of RF links that can do 35Mbps iperf3 rates.  This would likely need to be a 20MHz channel with distances less than a mile or two to achieve higher thoughput.   This looks to still be higher than the RF link and thus has no impact.

ai6bx, 2019-03-27:
The nodes I have loaded 3.19.3 to, 5 in all at the moment, appear to be steady. Nodes are set to auto negotiate and switches are still fixed 100 full. I can change those today and continue monitoring.

ai6bx, 2019-03-30:
My nodes have been stable since loading 3.19.3. We are looking at a pair of nodes with strong SNR yet less than desired overall throughput and will test by putting the patch in place to see if this has any effect. I suspect the larger problem is alignment but will use this as a test to be certain.

ae6xe, 2019-03-30:
There was a report in the forum that while the rates were still good and we
might consider the ethernet link stable, there was a reduced rate from
expected iperf ~80Mbps (on a 100Mbps ethernet protocol rate) dropping to
~35Mbps rate.  While this through put is still very good, some loss is
occurring.    Hopefully, this is not impacting the Elsinore to San Marcos
link, and we can check this off as a factor.

Joe

PR #493 (2020-03-23): bugfix: ubnt: pbe 400/400-iso/620 port hang
fixes #165
commit 7243fe08 (2020-03-23): bugfix: ubnt: pbe 400/400-iso/620 port hang

fixes #165
closed 2020-03-23 by ae6xe