aredn/aredn_ar71xx #166: Informational Note: Lat/Lon <> GridSquare Code

n2mh, 2018-08-29:
This is a link to code used by the weak-signal VHF/UHF/Microwave guys for converting between Lat/Lon and Grid Square. It was originally written in 1993 by W9IP and N1BWT and is in the public domain.

www.w1ghz.org/new/bd_2004.zip

It is written in pascal, so likely it will need to be ported to something else.

For use in a node, going from Lat/Lon to Grid Square would be useful for people who know their lat/lon already (or use the built-in map). This would be the obvious use.

Going from Gridsquare to Lat/Lon might be useful for people who do not wish to identify their exact lat/lon but are OK with an approximation, ie, the center of a 6-digit gridsquare. Folks in the aprs world do something similar. In the Mesh world, this could enable a node to be shown on a map without giving away the exact location.

Yes, in this situation, one could alter his lat/lon to gain the same effect. However, there is still the assumption that what was entered is an exact location. By entering only a grid square, the assumption now is that the lat/lon shown is only the center of a grid square and can be off by several miles from the true location.

No new feature is requested. Just something to think about with some enabling code.

r1de, 2018-09-03:
I'm sorry, I have to add my $.02 here from a mapping point of view...
No!

If you do not want to share your location then don't. (that is the "Opt Out").
"Fudging" the location this much is going to make some ugly looking maps...
The icons will stack up in the center of the grid square and in theory could be 10's to 100's of miles off...
That's not going to help anyone with planning or anything, it's better to have *no* location than this.

again, just my $.02. :smile:

dman776, 2018-09-03:
I agree.  That was my original design to allow the user to move the pin in the map if they didn’t want 100% accuracy but still wanted a pin.

Thanks,
Darryl

closed 2018-10-18 by dman776