aredn/aredn_ar71xx #171: Feature Polishing

n2mh, 2018-09-02:
There's been a number of new features added over the past several weeks. Nice job on all of them!

A few of them need a little polishing.

1. On the charts page, Signal to Noise Ratio (SNR) is shown as dBm. This should just be dB, not dBm. A ratio of two numbers is expressed in dB. When you add the m to dBm, this now means a ratio to 1 milliwatt. This turns the number into an absolute measurement. In this case, it would be called RSSI, the actual S-meter reading if you will. This is a measure of actual signal power.

The line just added refers to SNR. Thus, it is a ratio of signal power to noise power, dB. If the intent is to show actual received signal power, then the new line should read "Received Signal Power (RSSI)" or something like it. Then, you can keep the dBm unit of measure.

2. The OLSR Entries/Nodes statistic looks good. On the Mesh Status page, visually, I think it looks a little awkward. So, I put another horizontal line underneath it. IMHO the page looks better that way.

in /www/cgi-bin/mesh around line 323

`print "<table><th align='right'>OLSR Entries</th><td><nobr>Total = $olsrTotal<nobr></td></table>";`
`print "<hr><nobr>";   <===  add this line`

ae6xe, 2018-09-02:
1)  I think it should stay as SNR "dB" (not RSSI dBm).   Comes in handy when tunning the antenna position.   RSSI is less meaningful when the noise floor can float around, although is nearly always fixed at 95dBm.   Generally I think "RSSI" values should be avoided as the device is not very accurate in measuring absolute values.  It determines the noise level and then just says it is -95dBm and everything else is relative from there.

2) I agree, the additional spacing looks cleaner.

r1de, 2018-09-03:
How about just another `<br>` or 2.
Or under the buttons, off to one side of the page? inline with the buttons?
I am sorry, but, I think the extra `<hr>` is too much, it's clunky.
(I think it would look best somewhere else too, I tried a few spots but figured right in the center was a good starting point :) )

r1de, 2018-09-03:
Under Previous Neighbors would be better than where it is.
My Graphic Designer Wife says it would probably look best in the upper right corner, like this:
[image]

I like it up there too since it's sort of "telemetry" data and not the main focus.
(that's also where I originally had it, but thought no one would like it there but me :smile: )

Now I wonder what other info could be put there, free memory and /tmp space?
Whether Tunnels are enabled or not which prevents you from uploading new firmware?
etc...etc...

n2mh, 2018-09-03:
Actually, I wouldn't put anything there. Leave the area blank. I've seen some clubs put their club logo in that space.

r1de, 2018-09-03:
If they are adding a logo there, then they could bump the text down or take it away...
They'd obviously understand HTML, at least somewhat...  :smile:

closed 2018-09-06 by dman776

ae6xe, 2018-09-06:
https://github.com/aredn/aredn_ar71xx/pull/179

PR #177 (2018-09-06): aredn: Changed placement of OLSR info on the mesh status page.
Moved to be under the Previous Neighbors area.
It's actually under the previous neighbors listing
Fixes #171 

It's ugly where it is now, this is better, and eventually we'll find a good home for this info. :smile:
commit d9ce3179 (2018-09-06): aredn: Changeed placement of OLSR info on the mesh status page. (#177)

Moved to be under the Previous Neighbors area.
It's actually under the previous neighbors listing
Fixes #171