aredn/aredn_ar71xx #184: Potential Bug with Nightly 247

ai6bx, 2018-09-08:
I downloaded 247 ubnt-nano-m-xw this morning to load on a nano station M2 XW as a replacement for my current M2 XM. The load went fine, I replicated tunnel info and then moved my custom port forwarding scripts. The node booted and made an RF link with the node I was about to replace so all looked fine. I shut down the old node, activated Wan Gateway on the new node and took it up to plug it into port 1 on my tough switch expecting a DtD link to come live, the Wan gateway to come up and life to proceed better than it had been with tons of new memory in the node as it is the 64 mb model.

No Joy. there is no DtD link, and no gateway present. To test and see if there was something else awry I moved another node to port one and all seemed fine sans wan gateway. activating the gateway button on that node all worked. hmmm.

For now the node is powered down and the old node is live again.

ae6xe, 2018-09-08:
Keith,  I seem to recall, the NS M2 XW introduced an internal switch chip, like the NS M5 XW, and the function of the ports is different:

Primary:  LAN
Secondary:  DTDLINK & WAN

Orv, 2018-09-09:
Hopefully, as in the NSM5 XW, POE can be backfed from Secondary to Primary.  While not strictly recommended by AREDN, in a multi-node setup, running your network cable to the Secondary link of each XW-hardware based node will establish the customary communications.