aredn/aredn_ar71xx #186: Package Signature check failures

Orv, 2018-09-12:
NSM2 with nightly build 247-9ce76e1
Download Package Refresh failed with:

Downloading
http://downloads.arednmesh.org/snapshots/trunk/targets/ar71xx/generic/packages/P
ackages.gz
Updated list of available packages in /var/opkg-lists/aredn_core
Downloading
http://downloads.arednmesh.org/snapshots/trunk/targets/ar71xx/generic/packages/P
ackages.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/base/Packages.
gz
Updated list of available packages in /var/opkg-lists/aredn_base
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/base/Packages.
sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/arednpackages/
Packages.gz
Updated list of available packages in /var/opkg-lists/aredn_arednpackages
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/arednpackages/
Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/luci/Packages.
gz
Updated list of available packages in /var/opkg-lists/aredn_luci
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/luci/Packages.
sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/packages/Packa
ges.gz
Updated list of available packages in /var/opkg-lists/aredn_packages
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/packages/Packa
ges.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/routing/Packag
es.gz
Updated list of available packages in /var/opkg-lists/aredn_routing
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/routing/Packag
es.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/telephony/Pack
ages.gz
Updated list of available packages in /var/opkg-lists/aredn_telephony
Downloading
http://downloads.arednmesh.org/snapshots/trunk/packages/mips_24kc/telephony/Pack
ages.sig
Signature check failed.
Remove wrong Signature file.

dman776, 2018-09-12:
Build 256 is out

Thanks,
Darryl

Orv, 2018-09-12:
Ah!   OK - thanks.

closed 2018-09-12 by Orv