aredn/aredn_ar71xx #192: Expand Hostname Assignment field

Orv, 2018-09-15:
The Hostname data entry field appears to be about 15 characters wide (although hostnames can be up to 63 characters).   Unsure if the field could be that wide, but it would be nice if it were maybe 24-30 characters wide.
[image]

Orv, 2018-09-15:
Also notice that it shrinks the IP address field.  Unsure why that's happening, but if you're truncating the hostname, there's no need to shrink the IP address field.

Orv, 2018-09-16:
In thinking about it, the IP Address field needs only to be as wide as an IPV4 address - 15 characters (13 for a mesh IP address since it always starts with 10).  It could be narrowed to make room for a wider Hostname data entry field.

Orv, 2018-11-03:
I'm fine with that - we have unlimited space vertically on a page.

dman776, 2019-03-27:
I'm transferring this issue (since it is based on the Perl UI today) to the New AREDN UI (Angular) repo for capture