aredn/aredn_ar71xx #198: Add Refresh Feature to AREDN Online Map

k1ky, 2018-09-25:
Requesting a "Refresh" feature on the ONLINE AREDN Map which will retain the position, zoom level, etc. when viewing nodes.  This would be desirable when working with node updates to verify correct positioning and publication of node locations.  The current method of "reloading" a page requires the user to re-position and zoom in on the area of interest.

dman776, 2018-09-25:
Google Earth does this (and more) if you link the KML files

Thanks,
Darryl

k1ky, 2018-09-25:
Thanks for the tip Darryl, I do this with APRS.fi, didn't think about trying this with the AREDN map files. I'll give it a try.  I still believe my suggestion above would be a good enhancement.

k1ky, 2018-09-25:
The KG6WXC MeshMap product has the ability to fix coordinates and zoom level to the web address.  Would that be a viable solution/option?

dman776, 2018-09-25:
Also if I recall correctly you can include.  &lat=xxxx?lon=yyyy?z=n. On the url

For lat, lon, and zoom level

Thanks,
Darryl

k1ky, 2018-09-25:
What "format" are the coordinates? Zoom level parameters?  Can you provide a working example?

dman776, 2018-09-25:
https://www.arednmesh.org/content/31610-map-file-access

k1ky, 2018-09-25:
So as an Example for Nashville:
http://usercontent.arednmesh.org/K/5/K5DLQ/livemap2.html?zoom=3&lat=36.14&lon=-86.8&z=11
I can do a web browser reload and the map refreshes based on the provided coordinates and zoom level chosen.  That gets pretty close to what I was looking for.  If only the map would populate the browser addy with this information based on the current viewing position etc,. that would be perfect.

dman776, 2020-12-11:
@r1de fixed it!

closed 2020-12-11 by dman776