aredn/aredn_ar71xx #200: Routing locked to gateway even when access to the foreign network or internet is down

ae6xe, 2018-09-28:
When a foreign network is attached to the WAN of a mesh node,  routing will fail when this network is down:

1) when "mesh gateway" is checked, devices across the mesh will continue to attempt to route to this down network.
2) a device on the LAN network of the gateway node will also attempt to route to this down network, unless specifically preventing from accessing, if "disable default route" is checked.   

Since the foreign network is outside the control of the mesh gateway, there is no remedial steps the mesh node can take to resolve, however it would be possible to detect the foreign network is down by a config parameter that defines a pingable IP address or hostname on this network the mesh node could watchdog ping to ensure is available.   

If this address becomes unavailable, the address no longer responds, the node could stop advertising and routing to this foreign network and re-route traffic to the closest default gateway across the mesh network.

k1ky, 2018-09-28:
Excellent enhancement  - this is a much needed feature that will enhance network reliability especially during disaster situations.

ae6xe, 2018-09-29:
Changing this to a defect.   OLSR has a plugin implementing this capability and currently configured in the firmware.   A ping to the internet address occurs every 1 minute.

http://olsr.org/git/?p=olsrd.git;a=blob;f=lib/dyn_gw/README_DYN_GW

The current settings ping a hardcoded internet IP address.   If for example, the cat5 cable is unplugged from the WAN port (hAP lite or AirRouter), the default gateway route goes away from the main routing table.  But then, the node searches down the route table paths and finds another gateway on the mesh network.  The olsr watchdog ping succeeds, and OLSR continues to advertise that it's own gateway is still up.   

If anyone has other reproducible scenarios where this is not working, please post the specifics in a comment on this issue.

ae6xe, 2018-09-29:
Scenario 2:   When the path to the internet is down, and the cat5 continues to have the ethernet link 'up' on the WAN gateway, then the olsr watchdog ping will fail.   After several minutes, the mesh network is updated and no longer has awareness of the gateway.  This is working per design.    However, LAN devices of this mesh node have also lost internet access and are not switched to route across the mesh to find another gateway.    This may be possible to resolve (retain internet access across the mesh network)  by replacing ping with a traceroute, that must use the local gateway as the 1st hop to determine status.  Otherwise the 1st scenario failure would occur.

k1ky, 2018-09-29:
Having a method to "manually" trigger a test of these functions would be handy through the UI. Maybe even a green/red go/no go status indicator or something similar for the color challenged folks with regard to WAN Port internet availability and status?  Obviously don't want to over tax the node with these automatic checks, possibly a user settable monitoring interval?

ae6xe, 2018-09-29:
I could envision the node status page for the WAN section showing the possible states:

WAN Address:   <IP address>/None
My Mesh Gateway:   UP/Down/OFF
My LAN Device Access to my Gateway:  UP/Down/OFF

This is a bit wordy, but could be refined.  It's a bit challenging to convey all the possible combination of states.  

The implication when Down and OFF for "my mesh gateway", is that traffic would be routed across the mesh to another gateway.   The scope is traffic originating from other mesh nodes, not originating on the self node or local LAN. 

the implication when Off for LAN devices would be off (currently implemented this way "disable default route") for both local and remote gateways -- all or nothing to access the internet.    If 'down', the implication is traffic should be rerouted to a remote gateway.

n2mh, 2018-09-29:
Another possibility might be to ping over the WAN interface: `ping -I eth0.1` (that's an uppercase I) This might sidestep the scenario where a ping would be successful over the radio network.

However the ping gets done, the ip address to ping should be configurable.

ae6xe, 2018-09-29:
OLSR is currently using, "ping -c 1 -q <PING-ADDRESS>".   Adding the -I option would be straight forward to do and address scenario 1 bug.