aredn/aredn_ar71xx #204: A node will not communicate its name.  Neighbors display the IP address only.

ae6xe, 2018-09-29:
Ref:  https://www.arednmesh.org/content/occasional-dns-resolutionbroadcast-failures-new-release 

After booting, a node does not communicate its name to neighbors.    These neighbors will display an IP address only.    This functionality is part of the OLSR Nameservices plugin.      

If this situation is encountered after the node is booted, then boot again or restart olsrd - "/etc/init.d/olsrd restart".  This action is taken on the node with the IP address showing on its neighbors' mesh status.

Orv, 2018-09-30:
A reboot of an offending node cause the hostname to disappear from the OLSR broadcasts again - it seems pretty consistent.
Tried to restart olsrd and got this:

root@WD6EBY-LA-MtWilson-SW:/etc/init.d# ./olsrd restart
uci: Entry not found
uci: Entry not found
olsrd: ./olsrd: olsrd_setup_smartgw_rules() Warning: kmod-ipip is missing. SmartGateway will not work until you install it.
What's up with that?

ae6xe, 2018-09-30:
These errors/warnings have been showing for some time historically.  It's another to-do item on the long list.  But given no one sees any fallout, it is not very high priority on my list.

There must be a timing issue on the boot for olsr to have these symptoms.   This is helpful to know it is repeatable.

ae6xe, 2018-10-13:
Ref:  another symptom of this root cause:  
https://www.arednmesh.org/comment/9896#comment-9896

k1ky, 2019-01-03:
A bad side effect that I just noticed when a node name will not resolve at the far end, the SNR Chart function will only show the Average Signal for all stations and will not drill down to select a device even though there are several other "valid" nodes connected to it.  Also, the Archive feature goes to "revolving circle" on the web interface screen just like when there are no RF buddies connected as I reported in another issue here yesterday.  I have screen shots and support data available for this condition if needed.  Rocket M5XW running 3.18.9.0 as well as on Nightly 600 - no difference.

ae6xe, 2019-01-03:
did you have nightly 600+ on the node that is not propagating its hostname? Long shot, but just checking if cleaning up some config settings had an impact on these symptoms.

k1ky, 2019-01-03:
Yes, Installed Nightly 600 with same results.  Manual restart of olsrd did the trick.

ae6xe, 2019-01-07:
Root cause of this issue may also have symptoms reported in:

https://github.com/aredn/aredn_ar71xx/issues/323

One of the possible root causes is a timing issue when olsr starts up.  To test,   do the following one-time command (persists across reboots) to delay the startup of OLSR.  Does it make these symptoms go away?

mv /etc/rc.d/S65olsrd /etc/rc.d/S97olsrd

(Don't forget to un-do anything you added to /etc/init.d/local for this test.)

ae6xe, 2019-01-11:
This issue is believed to be resolved by https://github.com/aredn/aredn_packages/commit/4fe5e8eec06f7900b71a862c645ea374467c7fcd 

Please upgrade and do not edit in any work around to restart olsrd.   This fix is in build #620 as of Jan 11, 2019.   We'll let this ticket age to confirm the symptoms have gone away, then close accordingly.

ae6xe, 2019-01-12:
Does olsr restart in /etc/init.d/local prevent the condition as well?

Orv, 2019-01-12:
??  Not sure what that means, Joe.  /etc/init.d/local is a file, not a directory.

Orv, 2019-01-13:
And now I can't reproduce it.   Will keep testing...

Orv, 2019-01-13:
So - WD6EBY-LA-Verdugo-S-Sector-5G was upgraded to nightly build 620.  It's displaying the issue of IP address showing rather than hostname after reboot.  (Note - this node NEVER displayed this issue prior to upgrading to build 620).  Restarting olsrd resolves the issue.

In that node nowhere in /etc/init.d/local do I find a mention of olsr.

patrickdickey52761, 2019-02-03:
I can confirm that this happens on TP-Link nodes also. But after a while, if both nodes remain connected, it does resolve the hostname. I'm building the most recent version of the develop branch and will push the firmware out to all of my nodes. Then I can confirm if it's fixed or not.

byaeger, 2019-02-18:
Is this issue fixed? Ever since I flashed my Ubiquiti nanostaion loco m5 with this firmware it's been happening. Tried restarting olsrd from the cli and once restarted the neighbor disappears for about 20 seconds and then comes back only to show it's ip.

k1ky, 2019-03-03:
Issue still persists with Rocket M3 Nightly 737 today.  Need any more data or testing?

ae6xe, 2019-03-03:
Thanks, sufficient data is available.

AA7AU, 2019-04-12:
Similar issues with my AirRouter running 3.19.3.0 (and it seems [not certain] like it started after the upgrade from a much earlier Nightly Build).
See https://www.arednmesh.org/content/dns-over-tunneldtd-non-rf-node
AR was quiet and unattached overnight. Established a tunnel connection out to a CPE210 running 3.19.3.0 and connected fine as usual, however my AR only shows up as IP# on it, and when looking at another node on a DtD link to the CPE210, I see the 0.00 ETX ghosts as described in forum posts. Waited a couple minutes to let things settle in with no change.

Then ran /etc/init.d/olsrd restart over telnet on my AR and waited = no change,, same issues.

The ides of olsr timing being an issue might apply to me as my RF is turned off, which might affect the boot up timing and sequence. Also, once booted the RF IP# is not actualyl active as the radio is now being used as an AP.

HTH - Don - AA7AU

ae6xe, 2019-04-13:
Don, can you post support data downloads for the AR and the cpe210?   If the work around isn't working, and the symptoms are regularly repeatable, then let's see if we can find this root cause.

Turning RF off, just switches from a wireless to a sudo interface.   OLSR shouldn't be able to tell the difference, similar if there were no RF neighbors.

ae6xe, 2019-04-13:
On the AR, OLSR thinks the interfaces are down:
Table: Interfaces
Name	State	MTU	WLAN	Src-Adress	Mask	Dst-Adress
eth0.3975	DOWN
eth0.2	DOWN
tun50	DOWN
tun60	UP	1422	No	172.31.50.229	255.255.255.252	255.255.255.255

However, the interfaces are showing up:

eth0.2    Link encap:Ethernet  HWaddr 00:27:22:01:0D:E1  
          inet addr:10.1.13.225  Bcast:10.255.255.255  Mask:255.0.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:5443 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5446 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:511931 (499.9 KiB)  TX bytes:501178 (489.4 KiB)

eth0.3975 Link encap:Ethernet  HWaddr 00:27:22:01:0D:E1  
          inet addr:10.0.13.225  Bcast:10.255.255.255  Mask:255.0.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:689 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:89358 (87.2 KiB)

Do you have any network cables plugged into DTD or LAN ports?  If not, try putting a device on the LAN port and ensure there is a link light.  Restart olsr or reboot the node.  What happens?

AA7AU, 2019-04-14:
Ran exact same test again today, this time with ethernet wired LAN connection to AR instead of using WiFi LAN connection. Problem was persistent with WiFi (restarting OLSR did NOT help) - but symptoms do NOT appear to exist now with wired LAN connect - all looked good for first and second level abstracted connections.

Looks like you're right Joe, OLSR messes up how it sees the interfaces when there is not an active wired connection and RF is switched to AP mode.

Thanks - Don - AA7AU

mpannen1979, 2020-04-21:
No matter how many times I restarted olsrd, I could not get the hostname to push through.
Once I added a wired device to one of my LAN ports, it worked.

The hAP that I am using had nothing in the LAN or dtd ports to start, when this was happening.

ae6xe, 2020-04-21:
be sure to do the olsrd restart on the [IP address showing node].   Not on the node that is showing the IP address.

mpannen1979, 2020-04-21:
Yes, that's what I was doing.  Good reminder though.

mpannen1979, 2020-04-23:
If I go to either of the red circled nodes and trigger OLSR restart, the
node IP never gets resolved to a name back on this node.  Also, without a
name listed, it seems that other nodes outside of this node, don't get the
OLSR info from the red circled node, or any nodes downstream from the red
circled nodes.

Only if I have a device linked to a wired LAN Port on these red circled
hAP, does the OLSR restart effect.

I just wanted to clarify what I was seeing.

ae6xe, 2020-04-23:
"OLSR messes up how it sees the interfaces when there is not an active wired connection and RF is switched to AP mode."    Since this is a persistent issue, that I should be able to reproduce, it may be independent and a different defect from the intermittent cases when the OLSR restart is working.

mpannen1979, 2020-04-23:
I agree.  The workaround is doable for me (making sure I uplink something).  I wish I could provide some extra insight.  I will keep tinkering in the meantime.

mpannen1979, 2020-08-30:
May this will jog some thought.  Here is some netstat -l -p output from two nodes.  The first output is from a node that is having olsrd issues.  The second one is from a node that is not.  No matter how many times on the first node I restart olsrd, the name never propages, and the netstat output does not change.  This first node does not have any devices linked on the LAN ports at this time.  The second node has a few more attached olsrd PID:

[image]

mpannen1979, 2020-08-31:
Here is some debug output from olsrd startup.  KC9GMW is the node with OLSRD problems.
I don't know what any of this output means yet.

Can this help?

[olsrd-kb9oiv-1.txt](https://github.com/aredn/aredn_ar71xx/files/5153104/olsrd-kb9oiv-1.txt)
[olsrd-kc9gmw.txt](https://github.com/aredn/aredn_ar71xx/files/5153106/olsrd-kc9gmw.txt)

mpannen1979, 2020-08-31:
I can see that these lines from KB9OIV log never appear in KC9GMW:

Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Adding interface eth1.3975
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Adding interface eth1.2

mpannen1979, 2020-08-31:
E.G., here is the relevant section from the 'working' node:

Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]:  *** olsrd configuration ***
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Debug Level      : 9
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: IpVersion        : 4
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: No interfaces    : ALLOWED
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: TOS              : 0xc0
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: OlsrPort          : 698
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: RtTable          : 30
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: RtTableDefault   : 31
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: RtTableTunnel    : 254
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/all/rp_filter
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/eth1.3975/send_redirects
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/eth1.3975/rp_filter
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Adding interface eth1.3975
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/eth1.2/send_redirects
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/eth1.2/rp_filter
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Adding interface eth1.2
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Writing '0' (was 1) to /proc/sys/net/ipv4/conf/tun60/send_redirects
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/tun60/rp_filter
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Adding interface tun60

And here is the same section from the 'non working' node:

Mon Aug 31 23:10:11 2020 daemon.info olsrd[23968]: Writing '0' (was 0) to /proc/sys/net/ipv4/conf/tun60/rp_filter
Mon Aug 31 23:10:11 2020 daemon.info olsrd[23968]: Debug Level      : 9
Mon Aug 31 23:10:11 2020 daemon.info olsrd[23968]: IpVersion        : 4
Mon Aug 31 23:10:11 2020 daemon.info olsrd[23968]: No interfaces    : ALLOWED
Mon Aug 31 23:10:11 2020 daemon.info olsrd[23968]: Adding interface tun60
Mon Aug 31 23:10:11 2020 daemon.info olsrd[23968]: TOS              : 0xc0
Mon Aug 31 23:10:11 2020 daemon.info olsrd[23968]: OlsrPort          : 698
Mon Aug 31 23:10:11 2020 daemon.info olsrd[23968]: RtTable          : 30
Mon Aug 31 23:10:11 2020 daemon.info olsrd[23968]: RtTableDefault   : 31
Mon Aug 31 23:10:11 2020 daemon.info olsrd[23968]: RtTableTunnel    : 254
Mon Aug 31 23:10:11 2020 daemon.info olsrd[23968]: Willingness      : 3

ae6xe, 2020-08-31:
This data shows the start up details that OLSR knows about.   Both nodes have the same information known to olsr:

kc9gmw: Mon Aug 31 23:10:11 2020 daemon.info olsrd[23968]: Fixed Main IP: 10.214.123.237
kb9oiv: Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Fixed Main IP: 10.214.92.141

However, for the bad node, the hostname for this IP address, found in /etc/hosts, does not get propagate to other nodes.   it is a bit tricky to turn on olsr debug (tons of output) at the time the condition exists to capture.   Once you know there's a problem, OSLR is restarted in debug options, and then the symptoms may disappear.

mpannen1979, 2020-08-31:
The restart never fixes the problem.  It seems that unless something is 'linked' to a LAN port in this device, OLSRD never works properly.  Why would OLSRD not see the other 2 interfaces in this Mikrotik, as in the debug output above?  That might be indicative of something?  The subject devices does not have anything linked to an ethernet port.

ae6xe, 2020-08-31:
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Adding interface eth1.3975
Mon Aug 31 18:09:09 2020 daemon.info olsrd[14074]: Adding interface eth1.2

eth1.3975 shows only when the mesh RF is turned off.  This just means one node has mesh RF on, the other off.   eth1.2 will not show if the link is not up for a dtdlink connection.   There would need to be a cable plugged into the dtdlink port and connected to a switch or other live device.    Sadly, the detail in this debug info doesn't have sufficient information to trace the problem.

ae6xe, 2020-09-01:
It's been a while since I looked at this, but if there is a reproducible  combination, it would help to get the debug data and resolve.   But in the past, when I thought it was a combination of Mesh RF on/off and/or ether cable  plugged in or not, I wasn't able to reproduce reliably.    Maybe, I wasn't getting the right combinations, but if you can find 2 people that can reproduce the same combination of settings to see these symptoms, then let's capture the specifics and I'll try again.

mpannen1979, 2020-09-01:
OK, I was hoping this might help :(

We have problems on our local mesh where the Mikrotik devices are being used as wifi AP and tunnel clients, with no wired devices on the ports.  This causes problems with name resolution of nodes, since for some reason OLSRD is not propagating the name, regardless of how many times we restart the OLSRD service.

mpannen1979, 2020-09-01:
We have two devices now with this exact thing.  I only have root access to one of them at the moment.  There will be more of them in a few weeks, since we are having an 'AREDN' party this Saturday to give out Mikrotik's.

ae6xe, 2020-09-01:
I've seen turning mesh RF on or off seems to fix the problem for a given
instance, but It's been a while, don't recall specifics.  If you turn mesh
RF on and these symptoms go away, just turn the power down to the lowest
setting and a channel that will minimize any contention.

mpannen1979, 2020-09-01:
ahh ok, I will give that a shot, we aren't using the WAN wifi client option for the most part, so I will see what happens with that.

mpannen1979, 2020-09-01:
ok that seems to have allowed the name to propagate.  Thanks for the workaround suggestion!