aredn/aredn_ar71xx #205: iperf3 not available for current release

Orv, 2018-09-29:
On two separate nodes, both running 3.18.9.0, after refreshing the package list, and selecting iper3, got this when trying to install the package:

Unknown package 'iperf3'.
Collected errors:
 * opkg_install_cmd: Cannot install package iperf3.
Failed to restart all services, please reboot this node.

ae6xe, 2018-09-29:
I'm not able to reproduce.  I'll try some other nodes.  I get this when  doing a "refresh", selecting "iperf3" and isntalling (on an hAP AC Lite) below.  iperf3 was functional after install. 

Package Management
--
Installing iperf3 (3.5-1AREDN) to root... Downloading http://downloads.arednmesh.org/releases/3/18/3.18.9.0/packages/mips_24kc/base/ip erf3_3.5-1AREDN_mips_24kc.ipk Configuring iperf3. Failed to restart all services, please reboot this node. | Installing iperf3 (3.5-1AREDN) to root... Downloading http://downloads.arednmesh.org/releases/3/18/3.18.9.0/packages/mips_24kc/base/ip erf3_3.5-1AREDN_mips_24kc.ipk Configuring iperf3. Failed to restart all services, please reboot this node.
Installing iperf3 (3.5-1AREDN) to root... Downloading http://downloads.arednmesh.org/releases/3/18/3.18.9.0/packages/mips_24kc/base/ip erf3_3.5-1AREDN_mips_24kc.ipk 

Configuring iperf3. 
Failed to restart all services, please reboot this node.

Orv, 2018-09-30:
On a third node, I tried it and it worked.  I went back to the previous node and tried it again.  It found the package (image 1) but couldn't install it (image 2).  Will keep investigating,
[image]
[image]

Orv, 2018-09-30:
Well, it's like this - I had forgotten I had run those firewall Internet blocking commands you gave me, hence it did not have Internet access.  Once I rebooted that node I was able to download and install iperf3.

HOWEVER, on the adjacent DtD connected node (identical hardware and configuration) when I refreshed the package list, which was successful, iperf3 is missing.    I'll keep playing with it.

[image]

dman776, 2018-09-30:
i just installed iperf3 on a Nanostation M2 without issue.

Orv, 2018-09-30:
I'm 3 for 4; still researching the last one.  We can close this ticket until I find anything more substantial.

closed 2018-10-17 by dman776