aredn/aredn_ar71xx #206: Feature Request - Verify Node Board id before upgrading Firmware - match Firmware to appropriate Model for compatibility

k1ky, 2018-09-29:
Request that firmware compatibility to node make/model/board id type be checked before upgrading. If nothing more than a "warning" of possible incompatibility such as installing XM firmware on an XW device, mismatched Model i.e. installing Rocket Firmware on a Nanostation for example.

On several occasions even the most experienced user can accidentally install upgrade firmware on remote nodes, rendering them as "bricks", sometimes requiring a trip up a tower or a long trip to a remote site.

I can see situations where a total "lockout" for upgrading would not be preferable, but at least some sort of compatibility detection with a warning could avert imminent upgrade disaster.

ae6xe, 2018-09-30:
Been there, done that.

k1ky, 2018-10-01:
This could be accomplished by comparing for some sort of type match against the already installed firmware.  If different - provide a warning and confirmation to proceed.

dman776, 2018-10-02:
For a simple check, we could just verify the filename that the user is attempting to load.   ie:

<script>
	function validate(elem){
    var alphaExp = /.*-rocket-m-xw-sysupgrade.bin$/;
    if(elem.value.match(alphaExp)){
        return true;
    }else{
    	if (confirm('This filename is NOT appropriate for this device!  Are you sure you want to continue?')) {
 			return true;
		} else {
			elem.value="";
	        return false;
		}
    }
}
</script>
<input type="file" title="My File" name="myfile" accept=".bin" onchange="validate(this)" />

--------
This would mean that users would be warned if they renamed the files to something that would not comply with this check, but, they would still have the opportunity to override it by clicking CONTINUE.

PR #217 (2018-10-10): feature: Provide firmware file advice for uploading firmware files
also reduces page size by removing unnecessary style info on option tags.

fixes #206
commit 7a181793 (2018-10-10): feature: Provide firmware file advice for uploading firmware files (#217)

* feature: Provide firmware file advice for uploading firmware files

also reduces page size by removing unnecessary style info on option tags.

fixes #206

* update for Mikrotik filename exceptions

* correct download .list file name

* add support for cpe devices
closed 2018-10-10 by dman776