aredn/aredn_ar71xx #207: Slug Bug

ai6bx, 2018-10-02:
I suspect Slug Bug still exists. I have an AR and. nanoStation M2 XW that we’re prevouosly linked by DtD and RF to keep Slug Bug at bay. With the new release I broke the RF link and the Bug re-emerged witching 24 hours.

k1ky, 2018-10-02:
I agree - I have provided some support data on an AirRouter sans RF Buddy with high load values to Darryl.  Also experiencing high load creeping up on another AirRouter units with PBX and MapServer. Will try switching to 64MB units to see how that goes.

How do I attach a support data file here?  Github says it doesn't support that file type.

ae6xe, 2018-10-02:
These are actually different symptoms than slugbug.   This is listed in the release as a known issue, typically occurs with tunnel packages installed.  The difference is the node continues to function, and can show 5MB+ of free RAM and sustain a high load.  It is still working just slow.   Slugbug was the situation where the free RAM was on a downward spiral, until the node could no longer function.    Is the tunnel packages installed?  If so, what happens when you uninstall them.

Orv, 2018-10-02:
This is the normal load on the NSM5 that experienced high load earlier today ( > 5):
0.24, 0.25, 0.24.  Just for reference...

ae6xe, 2018-10-02:
Please monitor and capture with "uptime" and "free" data points.   We didn't see slugbug in the last couple of months of nightly builds before release, but that doesn't mean it won't appear.  It would be a matter of consuming enough RAM to occur.  The work done to address slugbug was to free up more RAM by removing unused features and non-critical packages like smpd.   I wouldn't recommend installing any add-on packages on 32MB RAM devices, they were at the ceiling in 3.16.x.x and more in 3.18.x.x.  

For lack of a better term, I'll call the new condition "heavy load".  It can be distinguished because the condition has been observed with upwards of 8MB RAM still free (25% of the RAM) and suspected of being coincidence with tunnel packages installed. 

Please validate the data points to confirm which scenario is occurring, with and w/o RF links.     Mitigate the issue regardless, 32MB RAM devices are pushing the ceiling on RAM and should avoid adding additional packages.

Orv, 2018-10-02:
We've seen at least a couple of occurrences locally.   The NSM5 I saw it on this morning does not, and never has had, tunnel software installed on it.

r1de, 2018-10-03:
I am seeing an NSM2 (with one tunnel client active) having a load of 1.54 at times, then it'll drop back down to normal, but it does seem to stay in a "high load" condition for longer it seems sometimes.

A few of  the times I have noticed it today, I was able to log into the node and run the "top" command.
Sometimes the top command will show a process using 80%+ of the CPU and then go away, the node seems to settle back down and then it happens again.
One of the process is: `{snrlog} /usr/bin/lua /usr/local/bin/snrlog`
That runs runs once a minute IIRC, so you would think the load would go up and down about every minute... It doesn't, sometimes that "snrlog" process seems to take a much much longer time than usual and holds onto the CPU.

I am *not* saying "I found the problem", this is just a pattern I noticed while watching today.

(I've also seen `{olsrd-namechang} /usr/bin/perl /usr/local/bin/olsrd-namechange` using quite a bit of CPU at times (70%), but it doesn't seem to get "stuck" like snrlog does)

Is there a good way to just disable snrlog?
I could dig thru the code (and will later), but it's been a long day today... :)

ae6xe, 2018-10-03:
There are multiple scripts kicked off each minute via crontab.  You could disable these, with loss of the respective capability.   We could change the freq of olsr updates that communicate information between nodes and reduce freq of updating dns when hosts come/go from the mesh.

Note, the bottleneck is not a 'cpu' bottleneck.  One can observe the cpu % utilization is not saturated during high load factors.    This turns the bottleneck to I/O--look for activity around I/O.   Need to find the specifics of this bottleneck.

r1de, 2018-10-03:
Now that you say that Joe, I have been watching the "STAT" column in top.
There is only ever one process (that I can see) that goes into the "D" state, and it is olsrd.
Another thing I notice is the vtund is "not nice".

Perhaps it is a combination of things coming into play occasionally...

I did stop snrlog, just for S&G and now, when I do see the load avg spike, it's not as much and never goes about about 0.8 for 5 seconds or so, then starts dropping back down.
It is during some of these spikes that I now notice olsrd in the "D" state...

(just more info, I'm still not sure what (if anything) this has to do with it, I'm just noticing things... ) :smile:

all the "state" codes are summarized here:
https://askubuntu.com/questions/360252/what-do-the-stat-column-values-in-ps-mean/360253#360253

Orv, 2018-10-09:
Found an NSM 5 (XM) not responsive.  Logged in and ran top, and found a high load (>5).   It dropped steadily to about 3, but when I tried to access the UI again it went back up to 5.  I stopped the browser, and it's settled down to about 2.8.  CPU utilization is nothing - 3%.

Don't know if it will be helpful or not,but I took a screenshot.
[image]

ae6xe, 2018-10-09:
yea, this is the symptom of "high load" and not "slugbug".  The node
continues to do its core function.    The free mem shows 5.8Mb, which is
plenty of space.   Don't know why it has high load factor, guessing it is
related to I/O bottleneck, since cpu is not fully consumed with this higher
load.     I see this on my NS M3 XM too.   But it floats between .8 and 2.0.

Joe

k1ky, 2018-10-09:
Ditto my observations regarding High Load values.  I just rebooted a Nanobridge M5 that has been running 8 days as a Tunnel Server (4 connections) and it was over 8.0 on the loading.  Recently upgraded to 3.18.9.0, it is connected to a Netgear GS108T managed switch.  This unit has never experience high load values running 3.17.0RC1 over many years.  I can obtain support data next time if needed to assist with this issue.  It has one VOIP phone attached talking to 3 PBX servers at remote locations on our system.

k1ky, 2018-10-09:
Back in the old days, we would increase the designated "swap space" on our Unix systems to help with stability issues, so much swap space for a particular number of users or hard disk size..  I am aware of the memory paring efforts, but obviously haven't hit on the right combination yet.  Is there a similar variable available in this environment to set aside more designated work area?

ae6xe, 2018-10-09:
These embedded devices don't have 'swap' space.   The only storage space outside of the 32MB RAM is the flash chip.  These chips are only 8MB and hold the persistent file system and kernel.   Even if there was space available on flash, it can only be written to ~100k times before physical failure.   All part of keeping costs low.  No physical options to configure and extend RAM or swap space.

k1ky, 2018-10-09:
Thanks for the background Joe - if I recall, the swap space was carved out of the available RAM on those systems.  We have solved issues with some RPi devices this way.

ae6xe, 2018-10-09:
Maybe the swap space was carved out of the SD card on the RPi, the equivalent is a mesh node's flash space (the low speed persistent memory).  We just don't have any space to carve out for this purpose.   

It would add a layer of inefficiency to reduce available RAM (the high speed memory), to use it as swap space.  It would/should all stay dedicated for available memory.

dman776, 2018-10-16:
on slow nodes with this condition, try this command to see if performance improves:
ulimit -s unlimited

ae6xe, 2018-10-16:
I also tested with "ulimit -l unlimited" in addition to the -s option.   This seemed to also make a difference and load values went below 1.0.   However, this is still not fully addressing the issue, as load factors were above 0.5.  Ultimately, a reboot of the node took the load values down below 0.4 (even after 3 days of runtime on my NSM3 XM).  Totally guessing, but this sure seems to be related to how linux manages RAM memory when it is limited -- on 32MB RAM devices.  I'm not optimistic that we would be able to do anything about this assuming inherit to the design of how linux manages memory or the 'iw' program implementation  'iw' program seems to be the center of the issues.  None of these symptoms are observed on 64MB RAM devices, so there can't be a memory leak somewhere.

k1ky, 2018-12-29:
What is the possibility of adding back the "feature" for OLSR to automatically restart when load values approach a certain high threshold?  This could increase the reliability (self-healing) of the system until this issue is resolved?  Seems like it would be a good practice that would reduce the need for manual intervention.

ae6xe, 2018-12-29:
There continues to be a watchdog feature for OLSR, such that if it is not performing-responding, it is automatically restarted.    Maybe this watchdog is already restarting OLSR when a node goes into this high load state, but I don't recall any examples.   It's not known if an olsr restart will fix the high load symptoms.   We can manually do a "/etc/init.d/olsrd restart" command to test.   

I suspect only a reboot will make the symptoms go away because the high load symptoms are continuing over a long period of time (days weeks)  past restart of most all the user processes.  The only thing that remains is the kernel.  One of the work around scripts looks for the "iw" process to be hanging and kills it to free up resources, but this also doesn't make the symptoms go away.

Are these conditions still occuring when "mesh rf" is disabled?

k1ky, 2018-12-29:
May need to check with Darryl on this but I believe the watchdog was disabled a while back as I have not seen any OLSR restart message counts in a long time.  It "appears" that the high load issue is resolved when RF is turned off.  I have an AirRouter HP running 6 active incoming tunnel connections stable for several days with 6 incoming tunnel clients with RF turned off where it wouldn't last but half of a day with RF on and no RF "buddy" attached.

dman776, 2018-12-29:
The OLSR watchdog process was removed and replaced by running oldrd under procd (from my recollection)

Thanks,
Darryl

k1ky, 2018-12-31:
What about adding a feature to disable all functions related to the Signal Graphing (Both Live and archive separately?) including the CRON processes for rssi and ID.  Other than OLSR, what else was changed from 3.16.2.0 that could contribute to the excessive loading??

ae6xe, 2018-12-31:
Tom,   3.16.x.x is based on Openwrt Barrier Breaker release in 2014.     These symptoms appeared in the following Openwrt release Chaos Calmer in 2015, and continues in the 2017, and 2018 releases.  The suspected area of the root cause is in the wireless functionality which comes upstream from Openwrt and others.    

There is only one place left, on a regular basis, that is calling the "iw" process.  This process is suspect and the center of things when we see the symptoms.    'iw" is called from the /usr/local/bin/rssi_monitor process every minute, data it needs is only available though 'iw'.  All the other places, snrlog, mesh status, node status, are all now calling "iwinfo" process and not believed to be contributing to the symptoms.

We can confirm by disabling this rssi_monitor.  Let's do this manually before implementing a feature to start turning things on/off, to know what will help or not.  The side effect is the node will become vulnerable to the deaf node condition, where it may tune out some neighbors as if noise, not recognizing the neighbor as a legit signals in favor of another signal.

To manually turn off this rssi_monitor (or snrlog),   "crontab -e"  and in the vi editor, comment out this rssi_monitor line (by putting a "#" at the beginning of the line).  I'm now doing this on one of my nodes that starts showing symptoms after 2 or 3 weeks.

r1de, 2018-12-31:
I have been commenting out the rssi_monitor cron job for a couple of weeks now (also the snrlog job), I thought I found a "root cause" of this high load issue since stopping one or both of those cron jobs did seem to help stop the nodes going into the high load scenario.
I tried to re-enable both rssi_monitor and snrlog on our networks problem node and the symptoms *did not* come back.
I wanted to see if I could turn it on and off so to speak, and I could not.
At that point the pattern I thought I saw broke down and I wasn't sure anymore, so I haven't said anything to very many people about it.
Since I read what your saying tho Joe, that might be the cause.

I've tried to catch when it happens and see what processes are running, and yeah, rssi_monitor is in there every time.
[image]

k1ky, 2019-01-01:
I did try disabling the Crontabs on those items and although it did seem to reduce the load on the node under test, it did eventually become unreachable.  I'll need to perform more testing.  One thing I've also noticed but not mentioned yet is that if I select the "Chart Archive" mode on a node with no RF neighbors, it just "hangs" there with a revolving circle and cannot exit - I have to use the browser "back" button to clear it.  not sure what that does to the Load, but it can't be good (IMHO).  Should I write up a separate ticket on that??

ae6xe, 2019-01-18:
Possible relationship to this issue:  
https://forum.openwrt.org/t/proposal-and-solution-for-high-load-fix-on-openwrt-luci/29006

dman776, 2019-01-18:
Yeah.   I saw that last week.   I forgot to mention it.

https://forum.openwrt.org/t/proposal-and-solution-for-high-load-fix-on-openwrt-luci/29006

k1ky, 2019-01-19:
Is this suggestion something that we can try by modifying a file or should we wait for a new Nightly release?  I'm willing to give it a shot on a problem N-620 node.

k1ky, 2019-01-19:
Does that line need to be '1' or just 1 ?

ae6xe, 2019-01-19:
to be persistent across a setup "save", put this  setting in /etc/config.mesh/uhttpd .   I'd keep consistent with other parameters, which are not quoted .

k1ky, 2019-01-21:
Experienced High Load on AirRouter with nightly 620 installed. Tunnel Client running with 1 outgoing connection.  Unit became very slow with load values exceeding 7 over a short period of time almost immediately after reboot.  No unusual activity and no IP phone attached.  Added the uhttpd patch with no improvement.  Removed VTUN module and re-installed - problem cleared up and unit now running loads near Zero. 

up time load average | 4:15 0.06, 0.13, 0.06
-- | --

closed 2019-02-13 by dman776