aredn/aredn_ar71xx #210: Auto Distance setting

ae6xe, 2018-10-03:
Add a setup option to auto configure distance dynamically in real time.

ae6xe, 2019-02-20:
Auto distance setting in 802.11 adhoc  mode with the ath9k wireless driver in use by AREDN was recently fixed and available in the nightly build code.  It only needs extended as an option into the UI to confirm it works.  See this openwrt commit:

https://github.com/openwrt/openwrt/commit/5ac34818f9c3ad46febc260f8ae348e05b9c8f00

This was tested on 2 x  Mikrotik BaseBox 5's 24km apart.

ae6xe, 2019-02-21:
After making some code changes,  I have successfully tested on a ~5 mile 3GHz link with P2MP condition.  The original author, didn't expect distances longer than ~24miles :) .

Here's what the logic of this auto distance is doing...    It is capturing the actual time taken for each 'ack' that comes back from each neighbor.   It then optimizes the ack time out value for the farthest out neighbor.   The ack timeout is the value computed and set from our distance setting and is for the node, not a value used for each station (although that would be ideal).    The time out value is captured-calculated:

ack_to =   (start time sending data over air) + (time remote node receives, processes, and ack's back) + (air time getting back all the ack bits)

An exponentially weighted moving average (EWMA) is used to smooth out and set the timeout value.

Max distance currently set to equivalent of 100km static setting.  Is this sufficient?  Anyone doing longer links for production?    I may have to increase this as I'm seeing higher than expected measured ack_to values, probably due to (time remote node receives, processes, and ack's back).

ae6xe, 2019-02-21:
For the early adapters and testers.   the auto distance setting can be tested:

limited time download available: 
https://drive.google.com/drive/folders/19K0tANbwzsbPhlEOSFZUX5osC0HevbQn?usp=sharing

Turn on auto distance:
"iw phy phy0 set distance auto"

Monitor Ack Timeout value set ("value  A" means Auto set,  "value S" means Static set):
"cat /sys/kernel/debug/ieee80211/phy0/ath9k/ack_to"

There is a iperf3 package at the download link that can be manually copied to the node and installed "opkg install <filename.pkg>".  You might need to "/etc/init.d/firewall restart" to open up the port.

k1ky, 2019-02-22:
Will we be able to "see" what the automatic setting is being set to in real time on the node??

ae6xe, 2019-02-22:
Type this command to "see" the setting:

"cat /sys/kernel/debug/ieee80211/phy0/ath9k/ack_to"

To convert from a static setting of Distance in meters to this value:

ack_to = distance/151.15151515  + 64

ae6xe, 2019-02-25:
closed by https://github.com/aredn/aredn_ar71xx/commit/d5be7814b34959c3ec8ad39080ff76eaf7571b8c

closed 2019-02-25 by ae6xe