aredn/aredn_ar71xx #211: Neighbor filter - black list

ae6xe, 2018-10-03:
1) prevent weak signal neighbors from constant up/down link associations
2) prevent part 97 violations

n2mh, 2018-10-13:
You might expand this to let the sysop create a Black or White List. The Black List would keep certain nodes out while the White List would only allow certain nodes in. I would also think that a White List would also be the same list needed for Issue 212 - Enabling only certain nodes for an event.

Enabling this feature could be dicey on a remote node. I would suggest a No List condition as the default option until specifically configured by the sysop. A confirmation question 'Are You Sure` would be in order before the configuration is applied.

The White List option harkens back to the NetRom days on packet when band openings would create short lived paths that were unusable for networking. They would disappear long before the routing algorithm knew they were gone.

n2mh, 2018-10-13:
I think Issue 212 is just another name for a White List. So, the answer is yes. How you want to call it would be different. I think White List/Black List would be more intuitive than Black List/Event Mode. Just my $0.02.

K6AH, 2019-01-31:
Ken, KE2N, suggests the black list is a standard feature in OpenWRT so its roots must be buried somewhere in AREDN.

K6AH, 2019-01-31:
Sorry about that...

Anyway K2EN says the black list is a standard feature in OpenWRT so its roots must be buried somewhere in AREDN.

ae6xe, 2019-01-31:
OpenWRT has a generic iptable rule builder to create a wide range of custom firewall rules.     In AREDN the rules can be added manually to the custom firewall rule table today.    However, in the spirit of AREDN,  we'd probably have UI sections for  specific use cases, e.g. a UI place to enter a list of MAC addresses to block, so that we can ensure the mesh node continues to operate properly and not break something else.

n2mh, 2019-02-07:
I think the best place to do this is via OLSR, not iptables. Done via olsr, the node(s) in question will never make it to the routing table. Thus, they don't have to filtered out downstream. Besides (I may be wrong here), once a node is in the OLSR table, it will still get propagated to other nodes. And, they will have to be filtered by iptables.

Some OLSR configuration statements to try:

Blacklist (per interface)
`LinkQualityMult 10.1.2.3 .1`  
where 10.1.2.3 is the ip address to be blacklisted
.1 is a multiplication value applied to the actual LQ coming in. Thus, an incoming LQ of 80% would be dropped down to 8% and used in subsequent OLSR calculations. If you set the multiplicatoion factor to 0, you have now fully blocked that ip address.

Whitelist (per interface)
`LinkQualityMult Default 0`  
Here you set anything coming in to 0, no quality. Then you use `LinkQualityMult` to enable those ip addresses you wish to accept.

Just a thought in the "more than one way to skin a cat" category.

K6AH, 2019-02-07:
Didn't we get strong objections around the White List concept at the recent
AREDN SoCal meet-up?

dman776, 2019-02-07:
Yes.  Users were strongly against it

Thanks,
Darryl

ae6xe, 2019-02-07:
move conversation related to white list rules to here:  https://github.com/aredn/aredn_ar71xx/issues/212