aredn/aredn_ar71xx #212: Event Mode -- lists of nodes authorized to communicate

ae6xe, 2018-10-03:
Ability to select a list of mesh nodes that may participate in an event.  Traffic from other mesh nodes would be blocked.   Selected on node by node basis for overlapping groups to limit communications.

ae6xe, 2019-02-07:
Users at a SoCal user meeting in 2018 voiced the following concern.    Owners of prominent backbone or cell tower sites using this capability may give preference to their own entity or someone else's.   The concern was a group being blocked from the network during an incidence or exercise.

virtualdxs, 2019-03-04:
I'm not a fan of this idea in general, mainly because I don't like the idea of the network having a "gatekeeper" so to speak deciding who does and doesn't get to play mesh in a given area.

xJMV, 2019-03-04:
It just depend on how it is used... Having the possibility to configure the whitelist while being disabled, this feature would allow a group to cut certain unnecessary traffic from the network in case of an emergency.

So all "mandatory" nodes are added to the whitelist while the filtering is OFF. So in a click or two, unnecessary traffic can be stopped in case of an emergency.

The blacklist option can also be used that way... it really just depends on the way a group decide to operate.

For me, more options = more possibilities.

xJMV, 2019-03-04:
Second thought: Whitelist could become a part in a QoS system, so priority goes to selected nodes without blocking any traffic...

virtualdxs, 2019-03-04:
The QoS idea I'd be in support of.

xJMV, 2019-08-17:
Another use about the whitelisting features would be for PtP link. A user in the middle of a PtP link can put up a node in the middle of the link and slowdown the traffic. Allowing only PtP devices would mitigate this problem and force users have their setup properly linked to the network.