aredn/aredn_ar71xx #215: hAP lite 5Ghz enable dual band capability

ae6xe, 2018-10-03:


dman776, 2018-10-19:
Suggest B) as a good starting point.  
A) would certainly require an additional page to define the firewall rules for port forwarding across the AP-mesh.

blm2008, 2018-10-19:
I like (B) as well and 29 would be plenty to get started with.  Would want to have a config page to manage SSID/WPA settings and maybe list wireless hosts.  A huge use case for this device for us is at a shelter or (public service event) aid station where we could offer network access to laptops or cell phones for VOIP, file access, and webcam viewing.  -Ben N4CV

Orv, 2018-10-19:
I'll put my vote in for B) also.
Orv W6BI

AC0WN, 2018-10-25:
I like "B" but I have some questions if I may:
I believe the suggestion is to use the 5 ghz radio as a configurable wifi access point.  If so, would the 2.4 ghz radio still be operating as an aredn node, or would it also be a wifi access point?  If the answer is the 2.4 is acting as an aredn node, could we add the ability to turn it off?
Any chance we can deliver +5 vdc to the usb port so it can be used as a handy power source?
Thank you to the entire AREDN development team for all the fine work!
73,
julie mcgrew
ac0wn

dman776, 2018-10-25:
Yes, the 2.4Ghz radio will be the "mesh" node side running OLSR.
The 5Ghz would be the access point (WPA2, etc).
Yes, the USB port is software switchable to supply 5VDC.

ae6xe, 2018-10-25:
There is another request in the pipeline to address the issue of "turn off the mesh RF":

https://github.com/aredn/aredn_ar71xx/issues/155

The capabilities on the hAP AC lite are being phased in over time.   In the ideal end state all combinations would be available to select,  both bands could be ether Mesh RF or Wifi AP.   This is with the kaveot that It is not yet known if we can make this 5GHz "802.11ac" device go into the part 97 channels, to be looked at.  It is a different driver than current driver in use for 802.11n wireless chips.  

Yes, it would be possible to turn power on the USB port.  I see the definitions exist in openwrt:
rb-952ui-5ac2nd)
	ucidef_add_gpio_switch "usb_power_switch" "USB Power Switch" "45" "1"
	ucidef_add_gpio_switch "poe_passthrough" "PoE Passthrough" "14"

For each of these incremental features we add, best to enter a standalone github  issue enhancement so we can track and close them out one at a time.

ae6xe, 2018-10-26:
No worries.  USB power on/off, just now created and see:  https://github.com/aredn/aredn_ar71xx/issues/241

PR #287 (2018-11-28): aredn: hAP ac lite enable 5GHz LAN Access Point
enable ability to bridge LAN physical ports with wireless
LAN Access Point capability. On dual band hAP ac lite,
used in parallel with mesh RF on 2GHz.  Enables future
features on single band devices to turn off mesh RF and
repurposed for mesh LAN Access Point.

closes: #215
commit a62fe95b (2018-11-28): aredn: hAP ac lite enable 5GHz LAN Access Point (#287)

* aredn: hAP ac lite enable 5GHz LAN Access Point

enable ability to bridge LAN physical ports with wireless
LAN Access Point capability. On dual band hAP ac lite,
used in parallel with mesh RF on 2GHz.  Enables future
features on single band devices to turn off mesh RF and
repurposed for mesh LAN Access Point.

closes: #215
closed 2018-11-28 by ae6xe