aredn/aredn_ar71xx #216: Avoid duplicate hostnames when reserving IP addresses

ae6xe, 2018-10-03:
Generic hostnames can be created across the mesh network, e.g. "RPi", "RasPBX", "ipCam", etc.,   If duplicate, one device may not be reachable.    Check if an existing hostname already exists for reserving an IP address.

r1de, 2018-10-05:
What about just a note, some text: "You are encouraged to prefix your hostnames with a callsign, be it a club or individual, this will ensure all hostnames are unique"

closed 2018-10-27 by ae6xe

AlyBadawy, 2019-01-10:
Thanks for this, and I see this was already patched. For the scenario above, but what about if the two nodes with such services are not linked until after the services names have already been setup?

ae6xe, 2019-01-13:
There's only an opportunity to test for duplicate hostnames after the fact when both nodes become live at the same time on a common mesh network.  We should be able to find the condition and report in mesh status to bring about awareness,

PR #226 (2018-10-27): bugfix: Check for existing hostnames on the network in DHCP reservations
fixes #216
*additional check for existing hostname when creating new DHCP lease.
checks hosts_olsr file if entered hostname already exists or not on the connected network(s).
outputs info saying such and gives advice to prefix hostname with callsign.
commit e8b3520e (2018-10-27): bugfix: Check to prevent duplicate mesh hostnames when reserving DHCP reservations (#226)

* bugfix: Check for existing hostnames on the network in DHCP reservations

fixes #216
*additional check for existing hostname when creating new DHCP lease.
checks hosts_olsr file if entered hostname already exists or not on the connected network(s).
outputs info saying such and gives advice to prefix hostname with callsign.

* change to use =~ /\s$host\S/i

* Reworded warning message

* moved the position of the hostname check to only when you click "add" on the DHCP leases page.

this stops it checking [... cut]