aredn/aredn_ar71xx #218: Reversion to 3.16.2.0 from 3.18.9.0 issues

k1ky, 2018-10-04:
I have some notes and discovered some issues regarding "reversion" from 3.18.9.0 and possibly newer Nightly and Dev releases back to 3.16.2.0:

1. The password does indeed change to "hsmm" (due to password encryption in newer releases)
2. The Time Zone is reset to UTC (this has been observed since 3.16.1.1 was released)
3. Publishing to the AREDN map appears to change the first 2 letters of the MAC addy so there will be 2 map entries, the old one and the new one as confirmed by moving the coordinates and republishing on 3.16.2.0.  (One is the LAN MAC and the other is the Wi-Fi MAC) probably should always publish the LAN MAC?

Suggest fixing the MAP publishing issue and the TimeZone issue if possible.

-----------------------------------------------------------------------------------------------------------------
I have developed a "downgrade" procedure:
0. Obtain good screen shots of all settings (just in case)

1. Remove Tunnel or other added modules like MeshChat if installed, reboot (no need to disable tunnel connections)
2. Verify WAN set to DHCP, change from STATIC if enabled, save & reboot if needed  (this will most certainly "brick" your node if not set to DHCP due to a data translation problem that was fixed later).
3. Install firmware 3.16.2.0 sysupgrade for appropriate model
4. Reset password from "hsmm" , reset Timezone to your preference, reset WAN to Static if needed. Save & Reboot
5. Reinstall tunnel (all settings should resume as before)
6. Republish to AREDN map

That should get it!

ae6xe, 2018-10-04:
The fixes are possible, but translates to another 3.16.x.x release.  Is this the intent/desire (taking away effort or pushing out other items on the to-do list)?

k1ky, 2018-10-04:
I wouldn't suggest slowing progress, but could be added to documentation/web site and considered for whenever a future 3.16.x update may be initiated. (If ever?)  I believe the map publishing issue has been fixed in the current release?  There is another ticket relating to my map suggestions relating to duplicates and displaying the most "recent" duplicate on top.

closed 2019-02-07 by dman776