aredn/aredn_ar71xx #221: DtD Link on CPE510v2

ke6scs, 2018-10-09:
I tried to get a dtd setup with two tp-link boxes, one was a CPE-210v2 and the other was a CPE-510v2, but they were not seeing each other, with or without a switch. After googling and fiddling with it for a while, what worked was changing some settings in the swconfig file on the CPE510.

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '5t 0t'

ke6scs, 2018-10-09:
This was observed using version 3.18.9.0

k1ky, 2018-10-09:
Did you check the 2nd port?  Those are normally configured DTD just like the Nanostation XW devices.??

ke6scs, 2018-10-09:
v2 devices only have one port

ae6xe, 2018-10-09:
KE6SCS, This would explain the issue. The cpe510 v2 is using the "cpe510-520-v1" image.  This image is built expecting the device to have 2 physical ports and an internal switch.   The cpe510 v2, with only 1 physical port, would not have an internal switch.  Let's get a comment from Andrew KK4ZUZ, previously looked at this device.

ae6xe, 2018-10-10:
I have gotten wan/lan/dtdlink all working on a single port on a NS M5 XW, with an internal switch.  This has not worked historically.  Good news...

Orv, 2018-10-10:
Joe, we've been using that configuration with NSM5 XWs for a couple of years now (and expecting it to break at any moment, but never happened).  It would be great if that could get officially sanctioned.

ke6scs, 2018-10-10:
Yes I can confirm that DtDlink and LAN are working properly with this tweak. I was previously having some trouble with DHCP on the release candidate, but that has cleared up with the GA. I have not tried to get WAN working as I have a TP-Link TL-SG105 which has its own little trouble with re-configuring VLAN 1