aredn/aredn_ar71xx #222: Reference to cdn.leafletjs.com

kd2hkb, 2018-10-12:
Ubiquiti Rocket M5 I'm getting two references to cdn.leafletjs.com for a stylesheet & a JavaScript file. This node is NOT connected to the internet nor does it have a WAN address.

kd2hkb, 2018-10-12:
[image]
[image]

ae6xe, 2018-10-12:
A node has a few internet addresses that it will attempt to reach out to for map details and time sync.   It is OK if it can't reach to them, this just means the functionality is not available to use -- node's time won't be set and some map details.   This information is not required or critical for core functions of the node.

kd2hkb, 2018-10-13:
Yeah, I'm well aware of that, but it does slow down the navigation of the administration pages to the point that even on a form submit... You wonder if it's going to fail or load.

ZL2ARN, 2018-10-13:
I have been going to comment on this issue for some time. No problems if you are connected to the Internet. However when you have a network of WIFI Mesh Nodes with NO internet connection, then you must suffer this 30 second (approx) delay every time you select not all but some of the pull down menus. For most uses not an issue. However for mesh sysops working remotely vi WIFI is gets very tedious.
Perhaps a shorter time, say 5 to 10 seconds would be more acceptable ??

n2mh, 2018-10-13:
Is this 30 second interval the normal dns timeout for an unsucessful host lookup? If so, there's probably a parameter somewhere that could be tweaked down somewhere. 
Some other ideas:
* Make the process of getting the map code happen only after the `Show Map` button is pushed. There's no sense to getting the map code if someone doesn't want/need the map.
* Ping an Internet address and if successful then fetch the map code. This ping might also be the same ping used to determine if the WAN port is good (Issue 200).

dman776, 2018-10-16:
duplicate of #223

closed 2018-10-16 by dman776