aredn/aredn_ar71xx #227: Log-in not active in 346 nightly builds

ZL2ARN, 2018-10-20:
Hi guys,
Many thanks for fixing the "30 sec delay in UI when no internet connection #222"
Now checking out build 346 on various node types I find that "log-in" is not active and the node is wide open to all to see and modify!!
Is this intentional for the nightly builds, or is this some obscure  result of issue "222 ??
I have reset my password on all affected nodes but to no avail.
There was a small delay previously before the log-in became active again. Perhaps this delay has been lengthened. I shall check again in 12 hours.
Gordon
ZL2ARN

dman776, 2018-10-20:
Clear you browser cache/history/sessions

Thanks,
Darryl

ZL2ARN, 2018-10-20:
Hi Darryl,
Definitely a Browser function.
Login to a Node, change to another Node, come back to the fist Node and still logged in.
Restart the Browser, then go to first Node again and yes login is required once more.
The Browser (Mozilla Firefox) cache is set to never save any pages. !!
Interesting browser behaviour.....

closed 2018-10-25 by dman776