aredn/aredn_ar71xx #230: Add ability in Settings to define a location for locally hosted images and packages

ae6xe, 2018-10-21:
Some mesh networks do not connect to the internet and are unable to 'refresh' in the Administration page to ease the ability to find and install newer firmware versions or install add-on packages.   Add a setting to enable definition of other locally hosted "download" web servers.    Instead of a  built in path of "http://downloads.arednmesh.org/releases/3/18/3.18.9.0/",  a group could clone these files and put locally on a web server at, e.g. "http://ae6xe-downloads.local.mesh/3.18.9.0".

dman776, 2018-10-22:
Questions on the functionality requested.
1) If a location is specified, should the node look EXCLUSIVELY in that location?
2) If not exclusive, should the node attempt the alternate location FIRST, then, the default location?
3) Should this setting persist across a sysupgrade, or, should the value "reset" to the default location in the firmware?

K6AH, 2018-10-23:
1.  I think the download location should be editable on the fly.  Also, there was some talk at the SoCal meeting about maybe specifying locations for both the AREDN Firmware and a separate for the packages.   In the end I think the admin should see all download options relevant to the device hosted at the specified URL.

dman776, 2018-10-23:
there will be a mechanism to go back to the default.
the edit's will be instantly available.  no reboot required.
I'd like to hear more on the reasoning/use case behind the firmware/package separation comment.

K6AH, 2018-10-23:
There's no need to specify a separate URL for the packages if they're at the same location.  I thought they were separate today.  The tough use case, I suppose is where you want to downgrade and still be able to load packages.

dman776, 2018-10-23:
Re: the downgrade comment. Packages have version specific locations embedded in firmware.  So, this should not be an issue.

PR #237 (2018-10-26): Add "Advanced config" page
Adds settings for leafletjs, leafletcss, maptiles, and firmware download location
fixes #230
commit 9391ff55 (2018-10-26): Add "Advanced config" page (#237)

* initial commit

* feature: advanced configuration page

fixes #230

* add help page updates

* added menu links to perlfunc.pm

* add default value for serverpath

* move "Reset to Firstboot" button from setup page to advancedconfig page

* secure page
minor ui changes

* center the help icon

* implement callback capabilities pre/post

* hardening uci calls to prevent command injection attacks

* moved settings to custom aredn uci file

* resolve shellcheck warnings
closed 2018-10-26 by dman776