aredn/aredn_ar71xx #231: Add descriptive field 'node type' and make available in json data file

ae6xe, 2018-10-21:
Add a field in basic setup to capture a node type, e.g. EOC, Backbone P2P link, 120 coverage, etc.   Extend field into http://localnode:8080/cgi-bin/sysinfo.json to enable network wide data collection for display on maps, understanding state of the mesh network, etc.

r1de, 2018-10-21:
I agree Darryl, that is what the description field is for, what I took away from yesterdays conversation was more of the ability to select a special icon for the device based on it's role in the network.
Something like a dropdown selection with: None, EOC, Police Station, Fire Station, Hospital, etc...
(I guess there could be others even, "No User Links", "Hilltop Node", "Offline", "Tunnels Only", etc...)
Selecting anything other than "None" would set special sysinfo.json field of "device_icon" (or whatever) and then when my map polling comes around polling the node, it could find this field and set that device to a different icon.

This was not the first time this request has come up, I've heard it before from some of the users of my mapping scripts, and it really only has to do with my map (so far).

I have this nearly done, I'm just trying to figure out how to create a new "system setting" in uci/nvram to store it in. :)

r1de, 2018-10-23:
That works for me, but is somewhat less intuitive than a simple drop down.
Up to you guys how you want to implement this, I can adjust the map code pretty easily (famous last words).

kylehamilton, 2018-10-29:
Would it be helpful to create a list of useful icon descriptions and sample icons? I can work some examples up over the next couple of weeks.

r1de, 2018-11-03:
After trying various things for a few days, I am leaning more towards Darryls solution.
The problem is "where does it end?", someone is always going to want another icon to be displayed.  
Even with a "tag" in the description field, it is the same problem, a possibly infinite number of additional icons and tags...
@kylehamilton send me a sample list of stuff you think we'd need and I'll see what I think...
(kg6wxc @ gmail)
Anyone else is welcome to submit the same idea to me as well... :smile:

dman776, 2018-11-05:
any objection to closing this pending the #tag solution above?

ae6xe, 2018-11-06:
None on my part.  the #tag in description field meets the need.

closed 2018-11-06 by dman776