aredn/aredn_ar71xx #233: Support for XW version of Bullet M2

jskier, 2018-10-21:
UBiquiti is selling Bullet M2 XW units.  Looking for a potential AREDN firmware.   I'm attaching DMESG output as well as board info file contents.  
[Bullet_M2_XW_board_info.txt](https://github.com/aredn/aredn_ar71xx/files/2499512/Bullet_M2_XW_board_info.txt)
[Bullet_M2_XW_DMESG.txt](https://github.com/aredn/aredn_ar71xx/files/2499513/Bullet_M2_XW_DMESG.txt)

ae6xe, 2018-10-21:
Looks like the loco-m-xw image will work on this device.  If you are willing to do this test, give it a try.  I would not expect a failed attempt (won't boot or respond) to be an issue, can directly tftp AirOS back on and carry on.   We're past the older issues of bootloader incompatibilities and soft-bricking.

The power settings won't be correct, consequently, assuming you can access after AREDN boots, then find this segment and edit in the file "/www/cgi-bin/perfunc.pm"

--------------from----------------------------
  '0xe866' => {
       'name'            => 'NanoStation M2 XW',
      'comment'         => '',
      'supported'       => '1',
      'maxpower'        => '22',
      'pwroffset'       => '6',
      'usechains'       => 1,
      'rfband'          => '2400',
    },
-----------------to---------------------------
  '0xe2c4' => {
      'name'            => 'Bullet M2 XW',
      'comment'         => 'Testing Bullet M2 XW',
      'supported'       => '-1',
      'maxpower'        => '19',
      'pwroffset'       => '6',
      'usechains'       => 0,
      'rfband'          => '2400',
    },
------------------------------------------------

Once this test occurs and is successful, this change and doc updates could go into a nightly build right away.

jskier, 2018-10-23:
Tried both 3.18.9.0-ubnt-loco-m-xw-factory. and 3.18.9.0-ubnt-nano-xw-factory images neither would load via UI as expected.  TFTP allowed load of images, but could not ping, ssh 20,2222, http to 192.168.1.1 using either image.  For grins tried restart with dhcp which as expected did not provide IP.     I was able to get u-boot backup of node and have expected Version: BAD, MAP: Good.  Ii have SSH access to the node, so if there are other files of interest I can grab them.

jskier, 2018-10-24:
The WIFI worked.  The setup says MAC2 in _setup.default is missing.  
Attaching dump and /etc/board.json  
[Bullet_M2_XW_board.json.txt](https://github.com/aredn/aredn_ar71xx/files/2508936/Bullet_M2_XW_board.json.txt)

[supportdata-NOCALL-201808160825.zip](https://github.com/aredn/aredn_ar71xx/files/2508945/supportdata-NOCALL-201808160825.zip)

The supportdata-NOCALL contains the TGZ file.

ae6xe, 2018-10-24:
The loco-m-xw image does not have the AR8035 Ethernet port driver in it for this device, so will not work.  We get this error at boot:

[    1.334007] ag71xx ag71xx.0: no PHY found with phy_mask=00000002

The rocket-m-xw image does have this driver.    Please tftp this factory image (or could upload the sysupgrade image over wifi in the Admin page).    This should get past this issue.

jskier, 2018-10-24:
info from the node.....

node: n5kke-2g-mobile3
model: Bullet M2 XW 

!!!! UNSUPPORTED DEVICE !!!!
boardid: 0xe2c4
Device has not been tested. Please file a ticket with your experiences.

eth0   FC:EC:DA:67:31:60
eth0.1 FC:EC:DA:67:31:60
eth0.2 FC:EC:DA:67:31:60
wlan0  FC:EC:DA:66:31:60
wlan0-1 FE-EC-DA-66-31-60-00-00-00-00-00-00-00-00-00-00

/proc/cpuinfo
system type		: Atheros AR9342 rev 2
machine			: Ubiquiti Rocket M XW
processor		: 0
cpu model		: MIPS 74Kc V4.12
BogoMIPS		: 266.64
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 32
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16 dsp dsp2
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='fc:ec:da:66:31:60'
hsmmmesh.settings.mac2='102.49.96'
hsmmmesh.settings.dtdmac='103.49.96'
hsmmmesh.settings.node='n5kke-2g-mobile3'
hsmmmesh.settings.config='mesh'

ae6xe, 2018-10-25:
The setting "'supported' => '-1',"  should provide an "Untested" message, not unsupported.  Check spelling or syntax, etc. of this section.     "'supported' => '1'," would not show any messages.    It is somehow getting a '0', which means the board id is not matching or other issue.  Can you upload a support download?  this might help identify.

jskier, 2018-10-25:
Joe,  The cgi-bin/sysinfo is reflecting the name element of the boardid entry.  I'm attaching the support dump.   I've verified the entry matches the '0xe2c4' provided earlier in the thread.  
Again thanks for your help.  
[supportdata-N5kke-2G-mobile3-201808160316.zip](https://github.com/aredn/aredn_ar71xx/files/2513155/supportdata-N5kke-2G-mobile3-201808160316.zip)

The perlfunc.pm entry is working correctly.

ae6xe, 2018-10-25:
"Untested" is supposed to show when the "supported = '-1'".   All looks good, so please take though the paces and post if anything comes up.  The sysinfo output is always showing "UNSUPPORTED DEVICE" until the "supported = '1'.   All looks good in the support download, nothing jumping out.

closed 2018-10-26 by dman776

ae6xe, 2018-11-19:
Jskier, can you contact me  ae6xe at arrl dot net or otherwise get me your email address?   For some reason the up stream openwrt team doesn't think one can "tftp" firmware to this Bullet M2 XW device.  They are operating with instructions that people have to pull the case and attached to the serial cover.   Yet we tftp'd firmware to this device.  Trying to figure out the gap.   

ref:  http://lists.infradead.org/pipermail/openwrt-devel/2018-November/014607.html 

can you run this command and send the output?    

cat /dev/mtd0 | grep -i u-boot

Joe AE6XE

k1ky, 2019-03-26:
The Rocket XW Firmware 3.19.3.0 loaded on a Buullet M2 XW device only allows up to 25dbm power setting.  This unit should support 28dbm.  Is this the right place for this comment or should there be another "new" ticket?

ae6xe, 2019-03-26:
The device's AirOS specs show it to only be max 25dBm.   Also the dmesg boot shows it has 64MB of RAM where as the specs say 32MB.  The specs must not be updated yet.

Joe

[    0.000000]  memory: 04000000 @ 00000000 (usable)  <- 64MB RAM

/var/etc/board.info from BULLET M2 XW  AIR OS  v6.1.5 
board.sysid=0xe2c4
board.cpurevision=0x00001122
board.name=Bullet M2
board.shortname=B2N
board.subtype=
board.netmodes=7
board.hwaddr=FCECDA663160
board.reboot=40
board.upgrade=150
board.phycount=1
board.phy.1.maxmtu=2024
board.phy.2.maxmtu=2024
radio.1.name=
radio.1.shortname=
radio.1.bus=ahb
radio.1.subsystemid=0xe2c4
radio.1.subvendorid=0x0777
radio.1.txpower.max=25
radio.1.txpower.min=-4
radio.1.txpower.offset=6
radio.1.ieee_modes=10
radio.1.antennas=1
radio.1.def_antenna=4
radio.1.antenna.1.id=4
radio.1.antenna.1.name=Main
radio.1.antenna.1.gain=0
radio.1.antenna.1.builtin=0
radio.1.antenna.1.chains=1
radio.1.antenna.1.chain.1.id=0
radio.1.antenna.1.chain.1.name=Chain0
radio.1.distance_limit_km=0
radio.1.chanbw="5,8,10,20,30,40"
feature.rssi.leds=4
feature.ext_reset=0

k1ky, 2019-03-26:
The specs I'm looking at show 28dbm for the Bullet M2HP.  They do have the memory specs wrong as published.  Also, what's with the 128mb specs that UBNT shows for some M5Ti and M2 Rocket models?
Also, is that 8MHZ channel bandwidth something that is available for us to implement?  Looking for some happy medium between 10 and 5 Mhz for better throughput below the 10Mhz connection.

PR #246 (2018-10-26): aredn: add definition for Bullet M2 XW
New Bullet M2 XW models are released by ubnt.

closes #233
commit 4a50ee23 (2018-10-26): aredn: add definition for Bullet M2 XW (#246)

New Bullet M2 XW models are released by ubnt.

closes #233