aredn/aredn_ar71xx #264: Auto wifi scan using javascript

raysuelzer, 2018-11-10:
Hey, 

I'm a front end web developer.  How would you feel about me taking a stab at implementing a javascript based autoscan feature that would update the page via ajax instead of meta-refreshes?  It would probably run faster and require less data over the air.

KM6WUH (soon to be KK6RAY)
Ray

ae6xe, 2018-11-10:
Ray,  YES.   please contact separately  ae6xe at arrl dot net.   We would love to rebuild a lot of the UI as the current perl generated pages are not very efficient, a memory hog, and don't get us started...  :) .  Let's further discuss.

dman776, 2018-11-16:
Reviewing this for a new UI approach with Ray.

dman776, 2019-03-26:
closing.   addressed with #379 and in aredn_ui PR#12

closed 2019-03-26 by dman776