aredn/aredn_ar71xx #279: Distance to FARTHEST Neighbor

ZL2ARN, 2018-11-23:
I find this entry on the Node Setup page just a little confusing, and it has taken me a year or two to fully understand exactly what the entry does.
The "help" file, for those of us who actually do read it, does not in my view, explain just what is required and how it works. 
This entry is particularly important with WIFI Networks where your home Node may see more than one remote Node. 
So am I right in thinking that we should set this entry for the closest Node that we can see from our home node. So that all packets go to this Node only , and then  they would be forwarded on from there.? 
I have been making these adjustments on our Wellington WIFI Network, and it does "appear" to speed up the responses from the distant Nodes further down the line.
So can we change **"Distance to FARTHEST Neighbor"** to better reflect exactly what is required here.  Perhaps **"Distance to FIRST Neighbor"**.
Gordon

ae6xe, 2018-11-23:
Gordon, Maybe it could be better defined as "Distance to the neighbor the traffic predominately routes though".     This is a time out value when a packet is transmitted, how long to wait for an ack back before we think it went lost -- and start sending the same data again, consuming more of the channel.   Distance to the first neighbor would give poor performance for all the neighbors farther out and take up more of the channel re-sending packets frequently.

Note, LQ/NLQ is not a good measure as these are broadcast packets with no ack back.   Measure a change in distance setting using iperf tests to see a direct correlation of how the link has been impacted.   With many active neighbors, choosing a fixed distance value is a lessor of the evils choice. 

There is another issue enhancement request to do an auto distance settings.

ZL2ARN, 2018-11-23:
Thanks Joe,
From my hole in the ground, I can see only the one neighbor node up the the adjacent hill top. And so I assumed that I would need only to set this short distance as the hill top node would be doing all the hard work linking further down the WIFI network.
However now when I set my distance to the farthrest node in the network, some 100km away, I can now get almost an instant response from those that were very marginal previously.
So perhaps as you suggest setting a reasonable distance in every node will indeed help to reduce the incidence of re-sending packets.
So back to "Distance to FARTHEST Neighbor". Now I understand the FARTHEST bit, but I do find the word Neighbor suggests the next door neighbor, and so tends to oppose  FARTHEST. !
How about   **"Distance to FARTHEST Network Node"**  ??

ae6xe, 2018-11-23:
Since this is a parameter on the 802.11 RF protocol, it is only applicable to nodes that are in RF range, that you are receiving signals from -- only the nodes that have a possibility of showing up in the Mesh Status Neighbor column.   If it is a hidden node or no chance of making a direct RF link, it would not apply.   If you receive signals from other hams connecting in to the same tower site, but never talk direct to or though them, then should be able to ignore them.    If you mostly or nearly always route traffic through a tower/cell site, then the distance to that site would likely be optimal.   Make sure the distance error is on the side of being too long.   Being too short means a time out before you get an ack back and a re-transmit.   How about, "Distance to farthest RF neighbor you are routing traffic to"?

dman776, 2018-11-23:
I suggest just updating the help file.  This new proposed description is a little wordy for the setup page form.  

Thanks,
Darryl

ZL2ARN, 2018-11-24:
This for me has been a misunderstanding of the usage for the word "Neighbor". I would define it as my immediate next door neighbor.  However now I see we have applied it to the entire group of WIFI nodes in a WIFI network. I can accept that is a good way to describe a cell of WIFI nodes in a local area. So when we tunnel through to  another remote cell of WIFI nodes , this would be another new neighbor group.
Perhaps as Darryl suggests we can further clarify the meaning of this in the HELP file.

ae6xe, 2018-11-29:
will add further clarity in the help.

PR #299 (2018-12-12): aredn: clarify distance setting in help
closes #279
commit 51340675 (2018-12-12): aredn: clarify distance setting in help (#299)

closes #279
closed 2018-12-12 by dman776