aredn/aredn_ar71xx #280: Unresponsive UI with recent nightly build

Orv, 2018-11-23:
Loaded aredn-465-f062021-ubnt-nano-m-xw-sysupgrade.bin on a test NSM5 XW.
Rebooted fine, but unable to load UI on browser - times out.  I  can ssh into system,and verified with nmap that uhttpd is listening on port 8080.
How to troubleshoot/recover?
Thanks.

k1ky, 2018-11-23:
Did you try connecting on the other port?

Orv, 2018-11-23:
Right after I posted this, a second power cycle was successful :-/   Let's close this for now; I'll keep an eye on it.

dman776, 2018-11-23:
Can you restart the web server?

/etc/init.d/uhttpd restart

Thanks,
Darryl

Orv, 2018-11-23:
I upgraded this node to aredn-471-2d5beb2-ubnt-nano-m-xw-sysupgrade.bin and rebooted it.  Same behavior: unresponsive UI.
Restarted the web server with /etc/init.d/uhttpd restart.  Same behavior - no web page returned. (verified it was listening on port 8080 with nmap).
Restarted uttpd AGAIN, and now UI responds.  Will continue watching node.

ae6xe, 2018-11-23:
I can reproduce.  Nothing I do is enabling the UI to work.  hmmm..   Anything in logread output related to httpd on your node?

ae6xe, 2018-11-29:
addressed in successive builds. reverted a change to uhttpd.

https://github.com/aredn/aredn_ar71xx/commit/557e4b3903ae0c320abce02557b42a9b1fdadf86
https://github.com/aredn/aredn_ar71xx/commit/483b138b3a7d76318ba9a934b294317fbd32e7b5

closed 2018-11-29 by ae6xe