aredn/aredn_ar71xx #288: CPE510 V3 Support

xJMV, 2018-11-29:
Hi,

I've just put my hands on 4 TP-Link CPE510 V3 device. V3 is actually supported, but I suspect it will be very similar to V2. 

Let me know what you need so I can provide information to help! This will be the first node on our network. I can actually SSH into busybox.

Here is dmesg:
`BusyBox v1.01 (2018.01.17-01:30+0000) Built-in shell (ash)
Enter 'help' for a list of built-in commands.

# dmesg
Linux version 2.6.31 (root@ubuntu) (gcc version 4.3.3 (GCC) ) #1 PREEMPT Wed Jan                                                                                         17 09:29:31 CST 2018
flash_size passed from bootloader = 0
CPU revision is: 0001974c (MIPS 74Kc)
ath_sys_frequency: cpu srif ddr srif cpu 560 ddr 450 ahb 225
Determined physical RAM map:
 memory: 02000000 @ 00000000 (usable)
User-defined physical RAM map:
 memory: 04000000 @ 00000000 (usable)
Zone PFN ranges:
  Normal   0x00000000 -> 0x00004000
Movable zone start PFN for each node
early_node_map[1] active PFN ranges
    0: 0x00000000 -> 0x00004000
On node 0 totalpages: 16384
free_area_init_node: node 0, pgdat 802429b0, node_mem_map 81000000
  Normal zone: 128 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 16256 pages, LIFO batch:3
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: console=ttyS0,115200 root=31:04 rootfstype=squashfs init=/i                                                                                        nit mtdparts=ath-nor0:128k(u-boot),64k(pation-table),64k(product-info),1536k(ker                                                                                        nel),6144k(rootfs),192k(config),64k(ART) mem=64M
PID hash table entries: 256 (order: 8, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Writing ErrCtl register=00000000
Readback ErrCtl register=00000000
Memory: 61376k/65536k available (1836k kernel code, 4088k reserved, 476k data, 1                                                                                        16k init, 0k highmem)
Hierarchical RCU implementation.
NR_IRQS:128
plat_time_init: plat time init done
Calibrating delay loop... 279.55 BogoMIPS (lpj=559104)
Mount-cache hash table entries: 512
NET: Registered protocol family 16
bio: create slab <bio-0> at 0
NET: Registered protocol family 2
IP route cache hash table entries: 1024 (order: 0, 4096 bytes)
Switched to high resolution mode on CPU 0
net_link: create socket ok.
TCP established hash table entries: 2048 (order: 2, 16384 bytes)
TCP bind hash table entries: 2048 (order: 1, 8192 bytes)
TCP: Hash tables configured (established 2048 bind 2048)
TCP reno registered
NET: Registered protocol family 1
ATH GPIOC major 0
squashfs: version 4.0 (2009/01/31) Phillip Lougher
NTFS driver 2.1.29 [Flags: R/O].
msgmni has been set to 120
alg: No test for lzma (lzma-generic)
alg: No test for stdrng (krng)
io scheduler noop registered
io scheduler anticipatory registered
io scheduler deadline registered
io scheduler cfq registered (default)
Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
serial8250.0: ttyS0 at MMIO 0xb8020000 (irq = 19) is a 16550A
console [ttyS0] enabled
PPP generic driver version 2.4.2
NET: Registered protocol family 24
7 cmdlinepart partitions found on MTD device ath-nor0
Creating 7 MTD partitions on "ath-nor0":
0x000000000000-0x000000020000 : "u-boot"
0x000000020000-0x000000030000 : "pation-table"
0x000000030000-0x000000040000 : "product-info"
0x000000040000-0x0000001c0000 : "kernel"
0x0000001c0000-0x0000007c0000 : "rootfs"
0x0000007c0000-0x0000007f0000 : "config"
0x0000007f0000-0x000000800000 : "ART"
TCP cubic registered
NET: Registered protocol family 17
802.1Q VLAN Support v1.8 Ben Greear <greearb@candelatech.com>
All bugs added by David S. Miller <davem@redhat.com>
athwdt_init: Registering WDT success
athwdt_timer_init: Starting WDT.
ath_otp_init: Registering OTP success
ath_clksw_init: Registering Clock Switch Interface success
VFS: Mounted root (squashfs filesystem) readonly on device 31:4.
Freeing unused kernel memory: 116k freed
xt_time: kernel timezone is -0000
nf_conntrack version 0.5.0 (1024 buckets, 30720 max)
ip_tables: (C) 2000-2006 Netfilter Core Team

Register vlan_manage hooks success.
PPPoL2TP kernel driver, V1.0
PPTP driver version 0.8.3
ATHR_GMAC: Length per segment 1536
ATHR_GMAC: fifo cfg 3 01f00140
mac:0 Registering S27....
ATHR_GMAC: RX TASKLET - Pkts per Intr:100
ATHR_GMAC: Mac address for unit 0:bfff0000
ATHR_GMAC: ff:ff:ff:ff:ff:ff
ATHR_GMAC: Max segments per packet :   1
ATHR_GMAC: Max tx descriptor count :   128
ATHR_GMAC: Max rx descriptor count :   128
ATHR_GMAC: Mac capability flags    :   2580
mac:1 Registering S27....
ATHR_GMAC: RX TASKLET - Pkts per Intr:100
ATHR_GMAC: Mac address for unit 1:bfff0006
ATHR_GMAC: ff:ff:ff:ff:ff:ff
ATHR_GMAC: Max segments per packet :   1
ATHR_GMAC: Max tx descriptor count :   128
ATHR_GMAC: Max rx descriptor count :   128
ATHR_GMAC: Mac capability flags    :   2D80
device eth0 entered promiscuous mode
device eth1 entered promiscuous mode
athr_gmac_ring_alloc Allocated 2048 at 0x83a6c000
athr_gmac_ring_alloc Allocated 2048 at 0x83a99800
WASP ----> S27 PHY MDIO
Setting Drop CRC Errors, Pause Frames and Length Error frames
Setting PHY...
ATHR_GMAC: Enet Unit:0 PHY:4 is UP RGMii 100Mbps full duplex
ATHR_GMAC: done cfg2 0x7115 ifctl 0x10000 miictrl
Setting Drop CRC Errors, Pause Frames and Length Error frames
athr_gmac_ring_alloc Allocated 2048 at 0x83b3f000
athr_gmac_ring_alloc Allocated 2048 at 0x83a97800
WASP ----> S27 PHY MDIO
ATHRS27: resetting s27
ATHRS27: s27 reset done
Setting Drop CRC Errors, Pause Frames and Length Error frames
Setting PHY...
br0: port 1(eth0) entering forwarding state

Disable VlanManage, data.enable(0), data.vid(0)
adf: module license 'Proprietary' taints kernel.
Disabling lock debugging due to kernel taint
ath_hal: 0.9.17.1 (AR9380, REGOPS_FUNC, WRITE_EEPROM, 11D)
ath_rate_atheros: Copyright (c) 2001-2005 Atheros Communications, Inc, All Right                                                                                        s Reserved
ath_spectral: Version 2.0.0
Copyright (c) 2005-2009 Atheros Communications, Inc. All Rights Reserved
SPECTRAL module built on Jan 17 2018 09:37:33
ath_dfs: Version 2.0.0
Copyright (c) 2005-2006 Atheros Communications, Inc. All Rights Reserved
ath_dev: Copyright (c) 2001-2007 Atheros Communications, Inc, All Rights Reserve                                                                                        d
ath_ahb: 9.5.2.18 (Atheros/multi-bss)
Enterprise mode: 0x00000000
Restoring Cal data from Flash
ar9300_convert_eeprom templateVersion = 82
===>>>[ar9300_eeprom_cover_ctl:5087] cover_ctl_flag: 0.
ath_get_caps[6209] rx chainmask mismatch actual 3 sc_chainmak 0
ath_get_caps[6184] tx chainmask mismatch actual 3 sc_chainmak 0
wifi0: Atheros 9340: mem=0xb8100000, irq=2
===>>>[ar9300_eeprom_cover_ctl:5087] cover_ctl_flag: 0.
dfs is NULL dfs_init_radar_filtersieee80211_wifipos_vattach: 3512: Wifipos intia                                                                                        lization starting
ieee80211_wifipos_vattach: 3525: Wifipos intialization mem alloc complete
ieee80211_wifipos_vattach:3539 Netlink socket created for wifipos:81cc3a00

ieee80211_wifipos_vattach: 3555: Wifipos intialization sock alloc complete
ieee80211_wifipos_vattach:3558 Registered WIFIPOS netlink family
ieee80211_wifipos_vattach: 3609: Wifipos intialization time alloc complete
ieee80211_wifipos_vattach: 3628: Wifipos intialization data store 1 alloc comple                                                                                        te
ieee80211_wifipos_vattach: 3638: Wifipos intialization data store 2 alloc comple                                                                                        te
ieee80211_wifipos_vattach: 3652: Wifipos intialization complete
VAP device ath0 created
dfs is NULL dfs_init_radar_filters
device ath0 entered promiscuous mode
br0: port 3(ath0) entering forwarding state
 Scan in progress.. Cancelling it
dfs_attach: use DFS enhancements
DFS min filter rssiThresh = 18
DFS max pulse dur = 151 ticks
dfs_attach disable detects
br0: port 3(ath0) entering disabled state
br0: port 3(ath0) entering forwarding state
Function ieee80211_acs_find_best_11na_centerchan best 5G channel is 48
ATH_RESET: call ath_internal_reset begin.
ATH_RESET: call ath_internal_reset end.
enet0 port4 down
ATHR_GMAC:unit 0: phy 4 not up carrier 1
br0: port 1(eth0) entering disabled state
enet0 port4 up 100Mbps Full duplex
ATHR_GMAC: Enet Unit:0 PHY:4 is UP RGMii 100Mbps full duplex
ATHR_GMAC: done cfg2 0x7115 ifctl 0x10000 miictrl
Setting Drop CRC Errors, Pause Frames and Length Error frames
br0: port 1(eth0) entering forwarding state
`

apcameron, 2018-11-29:
Does it have 1 Ethernet Port or 2?

Please also attach the output of the following shell commands

cat /proc/cmdline
cat /proc/cpuinfo
cat /proc/meminfo
cat /proc/devices
cat /proc/mtd
ifconfig -a
brctl show

xJMV, 2018-11-29:
Only one physical Ethernet port. I will come back with the rest of the information later today or tomorrow. :)

xJMV, 2018-11-30:
# cat /proc/cmdline
console=ttyS0,115200 root=31:04 rootfstype=squashfs init=/init mtdparts=ath-nor0:128k(u-boot),64k(pation-table),64k(product-info),1536k(kernel),6144k(rootfs),192k(config),64k(ART) mem=64M

# cat /proc/cpuinfo
system type             : Atheros AR934x
processor               : 0
cpu model               : MIPS 74Kc V4.12
BogoMIPS                : 279.55
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0000, 0x0428, 0x08f0, 0x08f8]
ASEs implemented        : mips16 dsp
shadow register sets    : 1
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available

# cat /proc/meminfo
MemTotal:          61564 kB
MemFree:           26652 kB
Buffers:            2716 kB
Cached:            10024 kB
SwapCached:            0 kB
Active:            11020 kB
Inactive:           9972 kB
Active(anon):       8252 kB
Inactive(anon):        0 kB
Active(file):       2768 kB
Inactive(file):     9972 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:          8272 kB
Mapped:             3200 kB
Slab:               9524 kB
SReclaimable:        444 kB
SUnreclaim:         9080 kB
PageTables:          236 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:       30780 kB
Committed_AS:      16064 kB
VmallocTotal:    1048404 kB
VmallocUsed:        2864 kB
VmallocChunk:    1037156 kB

# cat /proc/devices
Character devices:
  1 mem
  4 ttyS
  5 /dev/tty
  5 /dev/console
  5 /dev/ptmx
 10 misc
 77 ATH_GPIOC
 90 mtd
108 ppp
128 ptm
136 pts
238 ar7100_gpio_chrdev
239 ar7240_flash_chrdev

Block devices:
259 blkext
 31 mtdblock

# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00020000 00010000 "u-boot"
mtd1: 00010000 00010000 "pation-table"
mtd2: 00010000 00010000 "product-info"
mtd3: 00180000 00010000 "kernel"
mtd4: 00600000 00010000 "rootfs"
mtd5: 00030000 00010000 "config"
mtd6: 00010000 00010000 "ART"

# ifconfig -a
ath0      Link encap:Ethernet  HWaddr AC:84:C6:00:6C:A0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:28 errors:0 dropped:339 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:6972 (6.8 KiB)

br0       Link encap:Ethernet  HWaddr AC:84:C6:00:6C:A0
          inet addr:192.168.0.254  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1253 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1010 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:199681 (195.0 KiB)  TX bytes:555193 (542.1 KiB)

eth0      Link encap:Ethernet  HWaddr AC:84:C6:00:6C:A0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1278 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1010 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:900
          RX bytes:224581 (219.3 KiB)  TX bytes:560943 (547.7 KiB)

eth1      Link encap:Ethernet  HWaddr AC:84:C6:00:6C:A0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:900
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:16436  Metric:1
          RX packets:4184 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4184 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:282470 (275.8 KiB)  TX bytes:282470 (275.8 KiB)

wifi0     Link encap:UNSPEC  HWaddr AC-84-C6-00-6C-A0-00-46-00-00-00-00-00-00-00-00
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:34 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5902 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:2 Memory:b8100000-b811ffff

# brctl show
bridge name     bridge id               STP enabled     interfaces
br0             8000.ac84c6006ca0       no              eth0
                                                        eth1
                                                        ath0

apcameron, 2018-12-04:
We need to wait until TP-Link releases the firmware for this device.
https://www.tp-link.com/us/download/CPE510_V3.html
If I create  a build and it does not work you will not be able to recover without the firmware
It does look almost the same as the TP-Link CPE510 V2

xJMV, 2018-12-04:
Let's make some tests... Those 4devices will not be used anyway until AREDN works... Even if I have to recover one later it is not a problem so far...

I am well open to make tests next week when my work schedule is done.

xJMV, 2018-12-04:
By the way, it is available: https://www.tp-link.com/ca/download/CPE510_V3.html#Firmware :)

apcameron, 2018-12-05:
Unfortunately that link still only has the V 2 Firmware.

I will create a build for you to test in the next week or so and send an e-mail to you with a link to download the files from once it is ready.

dman776, 2018-12-05:
I thought TP-link used their version designators to differentiate HARDWARE level versions, not firmware level versions.   Ie.  Hardware for CPE210 V1.0 is DIFFERENT than CPE210 V2.0

apcameron, 2018-12-05:
Correct. That is why you should only use the Firmware version that is released for your Hardware version.
If you have a V3 device you should only use V3 Firmware or you may run into problems.
The link above is for Download for  the CPE510 V2 and not CPE510 V3 as the firmware has not been uploaded to their site yet for you to download.
If you look at https://www.tp-link.com/us/download/CPE510_V3.html there is no Firmware Tab yet.

xJMV, 2018-12-06:
@apcameron Weird I had it last week on the CA website... I just did not download it yet... was on my list, I remembered after I posted...

Sent an email to TP-Link to check if it can be obtained.

apcameron, 2018-12-07:
@xJMV I have sent you an e-mail with an image to test.

apcameron, 2018-12-07:
I found a site with the foreign version of the firmware https://www.tp-link.com/id/download/CPE510.html#Firmware
https://static.tp-link.com/2018/201801/20180126/CPE510(UN)_v3.0_20171228.zip

xJMV, 2018-12-07:
@apcameron I got it, will test monday when I get off from work! Will comeback asap to report! Thanks

xJMV, 2018-12-10:
@apcameron as requested, here is the support data.

Once installed with the factory image: https://paste.jmvien.net/?e5b9beca712c7eb0#QBS46zvOj7MtlC1PO+/gdIXaTjjGbk0s/g8akQ+PX0c=

Once updated with the sysupgrade image: https://paste.jmvien.net/?e4b59795c00a5690#DKpeV43jZd7Likc3/eJBD4AnMWfR5/JpJ9Eflo3vCyw=

5minutes into it only, I will start configuring everything and play with it, but the hardware runs it so far!

apcameron, 2018-12-10:
@xJMV That is good news. Once you have it fully configured please send me another copy of the supportdata so I can see if all the channels etc have been activated.
Also please try to run a sysupgrade via the Administration page again so we can validate that the sysupgrade process is working correctly.

xJMV, 2018-12-10:
So far it should be ready to deploy like this... However, when I do a WiFi scan, there is nothing that appears... 

https://paste.jmvien.net/?2bb05869349ae2d6#6mIbLYNp00aOv2NHsolxnCahZ9/LAY6nLgPsycjUBXg=

apcameron, 2018-12-10:
@xJMV If you do not have any other nodes configured it may not find anything on the Wifi Scan but @ae6xe or @dman776 would be the best ones to comment on that. 
I will go ahead and start working on a PR to include the TP-Link CPE510v3  in the build but will send you another built to test first  based on my PR

xJMV, 2018-12-10:
Roger that!

While waiting for a reply, I decided to open a second box to test further (I was already thinking about no results outside of other AREDN nodes) and I noticed that out of the 4devices I bought, I have 2xV2 and 2xV3.

So right now I am getting a V2 device ready for more tests! :)

@apcameron thanks for your time, I wait for the next test!

xJMV, 2018-12-10:
I can now confirm that both V2 and V3 sees each other :)

apcameron, 2018-12-10:
@xJMV As the V2  is already in the Nightly builds you will be able to download the files from there

apcameron, 2018-12-10:
Great. I will have a new build based on the PR I plan to submit ready in a couple of hours

apcameron, 2018-12-10:
@xJMV I have sent you an e-mail with the link to the final version to test before I submit my request to have it merged with the nightly builds.
Please test and let me know if it load successfully via the Administration Update option.

xJMV, 2018-12-10:
It does load and connect to V2 node.

You did not ask, but just in case : https://paste.jmvien.net/?d1d6caaeea0cc654#xDplau5M9+4yFMCM9nDP3ZYUCNPSa/bL29/XbVBFG/c=

apcameron, 2018-12-10:
That is good news. I will submit my PR to get it included in the Nightly builds

ae6xe, 2018-12-10:
https://github.com/aredn/aredn_ar71xx/commit/230eddf8bbd54bbe3dc99c5dfa5f2d5b1ee1734d

closed 2018-12-10 by ae6xe