aredn/aredn_ar71xx #294: AREDN Build #509

denis133, 2018-12-01:
Build #509 is working. I did notice that LED #5 (red) goes cold when the USR green LED goes hot but doesn't appear to have any impact on its operation.

ae6xe, 2018-12-04:
The default setting, and if you have specifically set POE passthough to be 'off', then this would be the expected and known behavior.   The hAP AC lite is powering up with the POE passthough on (red LED #5).   Once linux boots after about ~30 seconds, then the POE settings in AREDN are applied, so 5th light goes GREEN.     The USR light turns solid green at this same timing designating linux has booted.

denis133, 2018-12-04:
Okay but the passthough is set on and working, no led #5 however led does go hot green if a connected DtD. 
Thanks

ae6xe, 2018-12-04:
Oh, OK I miss-understood.  Another question, what device is the dtdport plugged into?    It is a proper load for POE to power?   It must be as #5 led is RED for a while ~30 seconds after boot.   Another optoin I need to check, on other devices, we have an available LED light to turn on when a mesh connection (RF or DtDlink) is established.   It might be turning on this light to green in error.

denis133, 2018-12-04:
It's connected to an AirRouter HP for testing with a 1 foot cable. 
This one is for your quirks post, as I type here I can't connect to the RouterBoard in question using Build 516 now from a working RouterBoard using 3.18.9.0 which is my main tunnel and Node. 
To confirm a bad connection I also tried the AirRouter HP RF connection to the Build #516 and can not connect. I then connected my laptop directly to the RouterBoard Build 516 and it is working perfectly using the 24v that came with it. I connected it back to the main node and it's back as normal!  All the above is RF 2.4GHz. The 5GHz has been working well. I am about to load the newest nightly build but I can hold off if you want me to check anything else?

denis133, 2018-12-05:
Negative that's why I didn't post it there but I wanted to mention it. I have run it with the PoE and original power both @ 24v. I can try it if you like.

denis133, 2018-12-06:
I ran this node on 12v and it performed without an issue. 
I updated it to build #522 and it would not connect to any node or see it. I connected it to a laptop and also could not see any link however I could link in over 5GHz AP. All settings looked normal.

closed 2018-12-09 by denis133