aredn/aredn_ar71xx #301: MikroTik hAP Lite fails to network over RF

n2mh, 2018-12-07:
I have two hAP Lites that I run on ch -2 @ 5 MHz. One is a US version and one is the INTL version. The INTL version runs nightly 491 and the US version runs nightly 516. The problem is this: both units do not network with Ubiquiti node on ch -2 @ 5 MHz. One is a Rocket about 20 feet away and the other is a NanoStation about 5 feet away. The Mikrotik's neither hear nor are heard by the Ubiquiti nodes. And, a WiFi scan didn't hear the Ubiquiti nodes either, including a different cluster of AirRouters on channel 9. However, they very happily network with each other.
Comparing notes with Denis, KD1HA, he reports a similar problem with one of his nodes, a US version. Another node he has is an INTL version and it works OK. Comparing serial numbers, the 3 nodes having problems all have a serial number starting with 924. The good one at KD1CA begins with 8B0.

Support dumps from both of my nodes is attached. Test1 is the US version and Test2 is the INTL version. Change .zip to .tgz. Gotta keep github happy!

[supportdata-N2MH-hAP-Test1-201812070325.zip](https://github.com/aredn/aredn_ar71xx/files/2655776/supportdata-N2MH-hAP-Test1-201812070325.zip)
[supportdata-N2MH-hAP-Test2-201812070326.zip](https://github.com/aredn/aredn_ar71xx/files/2655777/supportdata-N2MH-hAP-Test2-201812070326.zip)

ae6xe, 2018-12-08:
to confirm, all the tests and devices are running 24v?    

I'm not seeing anything pop out from the support data.  For lack of a better culprit, I'd say this is a hardware issue.

The channel width implementation is unique by the OpenWrt gurus.   Essentially, there is a change in a clock frequency to cause the channel to be cut in half.  Going from 20 to 10 to 5, is the same implementation except the relevant time values are changes.   symbol length is doubled, etc.   If it works from 20 to 10MHz, but does not going to 5MHz, this strongly suggests that there is a hardware limitation.  No sure there is anything we can do about this at this point.   

Let's compare the Qualcomm/Atheros chipset in this device.  Is it unique to any other device we have?  Possible there is a driver defect if this is a unique chipset -- it is "QCA9533 ver 2 rev 0"  If other devices are using this exact chipset and do not have this issue, then another indicator of a hardware failure.

dman776, 2018-12-08:
How is it powered exactly?  Mine is powered with a UBNT 24V PoE adapter into the POE IN port on the hAP AC Lite.  Not using external power jack.   Working fine with this configuration.

n2mh, 2018-12-08:
One is powered by the MikroTik wall wart and the other is powered by nominal 13.2 V from the shack power. And, both are powered through the barrrel connector on the front panel, No power is supplied via POE and no power is drawn anywhere either by POE or via the USB jack. I believe that if you swap power between the two units there is no difference in performance. I'll check that later today.

When I first powered up the first unit (the US  version), it seemed at the time that the unit would network when powered by 24V and not by 12V. However, that was fleeting and did not last. The present condition seems to be permanent.

I have two Ubiquiti devices on ch -2: one is a Rocket about 20 feet away through a couple of walls and the other is a NanoStation about 5 feet away from the MikroTiks. I also have a Bullet Titanium on the van in the driveway. I'll turn it on later today and see what happens there.

I also believe that Ian, KD6EPQ (new call AJ6??) was able to reproduce the problem. I'll let him comment here on what he found.

Would it help any if I opened up one of the boxes and look at the hardware inside?

n2mh, 2018-12-08:
Also, would you like a support dump from either of the Ubiquiti nodes to see what they are experiencing?

ae6xe, 2018-12-09:
Lower the power settings to ~50% power and retest.   Test the hypothesis that full power is distorting the signal in 5MHz.  Previous plots of 5MHz channels are not as square as 20MHz -- look more like a triangle in 5MHz.  Mikrotik's xmit power is generally higher on devices than Ubiquiti using the same wireless chipsets.   While this may give a marketing advantage, it may come at the cost of lower signal quality and higher out of channel energy.

n2mh, 2018-12-09:
Following up on my previous comments. Swapping the DC power between the units made no difference. The two MikroTiks only hear each other and still do not network with the Ubiquiti nodes. 
As a further test, I switched everything over to 10 MHz and all 4 nodes are now playing nicely in the sandbox.
Will switch everything back to 5 MHz and run your power test. QRX...

n2mh, 2018-12-09:
I dropped the power level on all 4 nodes down to their lowest setting possible (15 dBm for the NanoStation, 13 dBm for the Rocket, and 1 dBm for both hAP's. No difference in operation was found. Both Ubiquiti's hear each other and both MikroTiks hear each other. But MikroTik does not hear Ubiquiti and vice versa.

AJ6GZ, 2018-12-09:
I tested a hAP ac lite, Mikrotik BaseBox2, NSM2 XM and XW against each other at 5MHz.  Power lowered, all running at the same time, and various combinations of powered on/off. BaseBox2 will talk to hAP and the NSMs.  The hAP only meshes with the BaseBox2.  Running wi-fi scan from the hAP only shows the BB2.  All nodes within 20-50ft.  Tired various power levels on the hAP and even ran the distance up to 10mi just for kicks. Tested on build 522 then updated to 536, same behavior.
There is some uncertainty on the MikroTik forums as to which ac cards still actually support 5MHz like 'n' does.
Ian

ae6xe, 2018-12-09:
this is interesting if the BB2 can communicate to both the hAP and NSM2s, but the hAP and NSM2 are not linking in this 3 node adhoc setup.    Can you capture and upload the support data on all 3 devices during this test?  

The hAP has a QCA9531 802.11n chip for 2GHz and a QCA9887 802.11ac chip for 5GHZ.   The BaseBox 2 has an AR9342.    The NSM2 XM is older AR7240.  The NSM2 XW is newer AR93xx, maybe exact one as BB2.

This 3 node link results rules out things like the hAP on 5MHz has shifted center frequency.   Power levels and possible impact on signal distortion isn't making a difference.   The support data will indicate if there is a 802.11g/n capability miss-match to rule out.

dman776, 2018-12-12:
Wondering if there is any correlation between serial numbers.....
My "working" hap aclite serial number is:
7C34085E6880/746

n2mh, 2018-12-12:
One other thing to throw into the mix.

We may be looking at a temperature problem. On a hunch, I took one of the units and put it into my non-heated garage. It's in the '40s there right now. And, it now sees the NanoStation in the shack at `82% | 90% | 20.6` and a Bullet Titanium that I just powered up outside on the van in the driveway at `100% | 90% | 16.2`. Interesting that it still doesn't see the Rocket and the link quality to the other hAP is down `36% | 94% | 28.9`. It's been out there for a couple of hours now. Power is 24V from the supplied wall wart.

I wonder what putting it inside the refrigerator freezer would do ...

n2mh, 2018-12-22:
Well, that didn't last long. After a few hours, it reverted back to a no networking status. So, I disconnected it from power, let it sit for a day or so in the cold garage and then powered it back up. It lasted a little bit longer this time, but eventually stopped networking.

At this point, I'm thinking this is a hardware problem, fixibility uncertain.

AJ6GZ, 2018-12-23:
[4-node-5Mhz.zip](https://github.com/aredn/aredn_ar71xx/files/2706411/4-node-5Mhz.zip)
Here is the support data from a second test again with the 4 nodes.  This time only the two MikroTik seemed to connect, though the XM NSM2 tried but never gained a high enough LQ/NLQ to be useful.  Also performance on both connected sets suffered likely due to cross interference of course.  I didn't roll back power on anything this time.

AJ6GZ, 2018-12-24:
[4-node-5MHz-lowered-power.zip](https://github.com/aredn/aredn_ar71xx/files/2706417/4-node-5MHz-lowered-power.zip)
And again the same test with power set at 19dBm across the board.  The BB2, normally high powered on 10Mhz and works fine, connects to all.  hAP only connects to BB2 as seen a few weeks ago and traffic works ok.  So the BB2 is obviously overloaded at 30dBm/5Mhz, probably never designed for this.
I also lowered everything to 12dBm with the same results and also the hAP only at 6dBm with same results.

ae6xe, 2019-08-16:
I tested 2 BaseBox 5 devices from NH6HI and have reproduced this failure.  The BaseBox 5 devices, while I could get them to link together, would not link with a Ubiquiti NanoStation M5 XW.  When changing from 5MHz to 10Mhz channel widths, everything worked.    

This symptoms are not limited to Mikrotik hAP ac lite devices, now Basebox 5 devices -- model 912UAG-5HPnD-OUT.