aredn/aredn_ar71xx #319: Minor GUI Display Issue

Orv, 2018-12-30:
Using Internet Explorer 11, there's a blue rectangle around the AREDN logo on the firmware status page.

Using Firefox version 64.0 on both Windows 10 and Fedora Linux and Google Chrome 71.0.3578.98 on Linux, that blue rectangle is absent.

ae6xe, 2019-01-06:
We can add something like this to the html-css code to prevent this:  style=”border-style: none .

dman776, 2019-01-09:
Orv, can you try this css change (I don't have IE 11).
edit /www/aredn.css
change this line:
.AREDNLogo { background:rgb(231,231,231);  position:absolute; left:10%; width:175; height:95; }
to read:
.AREDNLogo { background:rgb(231,231,231);  position:absolute; left:10%; width:175; height:95; border-style: none;}

ae6xe, 2019-01-09:
I added this change.   started IE 11, cleared browser cache for everything.  browsed to the status page and did not see the blue rectangle box anymore.

Orv, 2019-01-09:
For me, this change resolves the issue with IE 11, for both URLs Joe mentioned.
NOTE - I only saw the blue rectangle on my Mikrotik HAP ac Lite; it does not appear on any of my Ubiquiti devices.

ae6xe, 2019-09-04:
Orv, do you want to confirm this was resolved before we close this one?  Fix was submitted and in build a while back.

Orv, 2019-09-04:
Soon - still traveling.

Orv, 2019-09-07:
It appears to be resolved, and may be closed.

closed 2020-04-22 by ae6xe