aredn/aredn_ar71xx #321: SNR Chart Archive Hangs with NO RF Buddy

k1ky, 2019-01-01:
The "Chart Archive" mode  "hangs" there with a revolving circle and cannot exit - I have to use the browser "back" button to clear it.  This condition exists on nodes that have no RF Connections. Suggest fix as well as provide an option for the user to totally disable the SNR Logging system along with related Crontabs processes when no RF is present with the RF deck enabled.  I see that the option has been removed when the MESH RF is disabled.  Could this be contributing to the excessive "load" issue?

k1ky, 2019-01-15:
I have an AirRouter running dev-176 operating as a Tunnel Server, no RF Buddy and phone attached that runs with low load values and does not "Hang" when selecting the archive.

ae6xe, 2019-03-03:
possibly related:  https://github.com/aredn/aredn_ar71xx/issues/204#issuecomment-451033842

k1ky, 2019-03-04:
Joe - well - in a way yes in that this condition (#204) represents an "unknown" connection and the process hangs and doesn't allow selection display the other available valid connections either in the Archive or the "LIVE" Real-Time connection selection area even though they are connected and working fine.